raspberrypi/usbboot #1: Buildroot.elf still obsolete

ED6E0F17, 2016-11-22:
Continuation of https://github.com/raspberrypi/tools/issues/44 : buildroot.elf is still dated Oct 28 2014

It was suggested that buildroot.elf would be replaced with the standard start.elf from firmware-next, so that the Pi Zero could boot as a USB Device without an SD card, in the same way that it can now boot as a USB  Host with only bootcode.bin on an SD card.

ghollingworth, 2017-01-20:
You can now use the start.elf from the standard firmware (although at time of writing you'll need to use the next branch of the firmware to get USB device support)

closed 2017-01-20 by ghollingworth

ED6E0F17, 2017-01-20:
I was about to close that, but I have not successfully tested it on PiZero yet.