raspberrypi/usbboot #8: I have to run the tool twice to make it go in storage mode

agherzan, 2017-01-20:
```
andrei@carbonA rpi-usbboot $ sudo ./rpiboot 
Waiting for BCM2835/6/7
Sending bootcode.bin
Successful read 4 bytes
```

Nothing detected.

```
andrei@carbonA rpi-usbboot $ sudo ./rpiboot 
Waiting for BCM2835/6/7
Second stage boot server
File read: start.elf
```

Correctly detected.

ghollingworth, 2017-01-20:
Ah, we run around the loop twice when we program the CM3 so turning off -l (loop) will make this happen...

If you add the -l command line option then it'll work but it just won't exit after programming...  Will have a think about it..

Gordon

agherzan, 2017-01-20:
At least this needs to be communicated in [documentation](https://www.raspberrypi.org/documentation/hardware/computemodule/cm-emmc-flashing.md) .

ED6E0F17, 2017-01-20:
I suspected that you might want "while(loop--)" , with the -l option set when -d.

tvoverbeek, 2017-01-20:
See my pull request #9. Should solve it.

ED6E0F17, 2017-01-21:
I need to loop three times for a file-system boot (option -d). Your patch does not fix this for some reason.

tvoverbeek, 2017-01-22:
I assume a file-system boot relies on the right combination of bootcode.bin, start*.elf, fixup*.dat and kernel.img.
When I try with an RPi as host using its /boot (firmware updated today with rpi_update) as file-system directory I have to put the bootcode.bin from the usbboot/msd directory in /boot from the RPi host.
Then booting a PiZero with 'sudo ./rpiboot -v -d /boot' I get the read of bootcode.bin, config.txt, start_x.elf and fixup_x.dat.
Then silence, no Done packet and rpiboot is waiting.
Eventually the green PiZero led flashes 7 times (kernel.img not found).

How does the kernel get loaded in usb-device boot ????
@ED6E0F17 how is your file-system boot working ???

ghollingworth, 2017-01-22:
You need to use all the files from the next branch of the firmware.  I think you're using a start.elf that does not support USB device mode...

tvoverbeek, 2017-01-23:
With the latest firmware from the next branch (commit 7c1a883) device boot on a PiZero works, but I have to run usbboot twice (or use the -l option).
First time through it exits after loading start.elf.
On the second run it finds device idVendor=0x0a5c idProduct=0x2764 with serialnr 1 and goes to the second stage fileserver. Then it loads the remainder.
Here the output from 'sudo ./rpiboot -v -l -d /boot' with the latest firmware in /boot on a RPi 3:
```
Waiting for BCM2835/6/7      <--- First run
Found device 1 idVendor=0x0a5c idProduct=0x2763
Device located successfully
Initialised device correctly
Found serial number 0
Sending bootcode.bin
libusb_bulk_transfer returned 0
Writing 50268 bytes
libusb_bulk_transfer returned 0
Successful read 4 bytes 
Waiting for BCM2835/6/7      <--- Second run
Found device 1 idVendor=0x0a5c idProduct=0x2764
Device located successfully
Initialised device correctly
Found serial number 1
Second stage boot server
Received message GetFileSize: autoboot.txt
Cannot open file autoboot.txt
Received message GetFileSize: config.txt
File size = 1647 bytes
Received message ReadFile: config.txt
File read: config.txt
libusb_bulk_transfer returned 0
Received message GetFileSize: recovery.elf
Cannot open file recovery.elf
Received message GetFileSize: start.elf
File size = 2833636 bytes
Received message ReadFile: start.elf
File read: start.elf
libusb_bulk_transfer returned 0
Received message GetFileSize: 
Second stage boot server done
Waiting for BCM2835/6/7
Found device 1 idVendor=0x0a5c idProduct=0x2764
Device located successfully
Initialised device correctly
Found serial number 1
Second stage boot server
Received message GetFileSize: recovery.elf
Cannot open file recovery.elf
Received message GetFileSize: config.txt
File size = 1647 bytes
Received message ReadFile: config.txt
File read: config.txt
libusb_bulk_transfer returned 0
Received message GetFileSize: dt-blob.bin
Cannot open file dt-blob.bin
Received message GetFileSize: recovery.elf
Cannot open file recovery.elf
Received message GetFileSize: config.txt
File size = 1647 bytes
Received message ReadFile: config.txt
File read: config.txt
libusb_bulk_transfer returned 0
Received message GetFileSize: bootcfg.txt
Cannot open file bootcfg.txt
Received message GetFileSize: cmdline.txt
File size = 136 bytes
Received message ReadFile: cmdline.txt
File read: cmdline.txt
libusb_bulk_transfer returned 0
Received message GetFileSize: recovery.img
Cannot open file recovery.img
Received message GetFileSize: kernel.img
File size = 4345528 bytes
Received message GetFileSize: armstub.bin
Cannot open file armstub.bin
Received message GetFileSize: kernel.img
File size = 4345528 bytes
Received message ReadFile: kernel.img
File read: kernel.img
libusb_bulk_transfer returned 0
Received message GetFileSize: bcm2708-rpi-b-plus.dtb
File size = 13168 bytes
Received message ReadFile: bcm2708-rpi-b-plus.dtb
File read: bcm2708-rpi-b-plus.dtb
libusb_bulk_transfer returned 0
Received message GetFileSize: config.txt
File size = 1647 bytes
Received message ReadFile: config.txt
File read: config.txt
libusb_bulk_transfer returned 0
Received message Done: config.txt
Second stage boot server done
Waiting for BCM2835/6/7      <--- Third Run
Found device 1 idVendor=0x0424 idProduct=0xec00
Found device 2 idVendor=0x0424 idProduct=0x9514
Found device 3 idVendor=0x1d6b idProduct=0x0002
Found device 1 idVendor=0x0424 idProduct=0xec00
Found device 2 idVendor=0x0424 idProduct=0x9514
Found device 3 idVendor=0x1d6b idProduct=0x0002
```
Should the logic be changed to run the file-server twice in case of a file-system (-d) boot?
Or just leave as is and document it?

ghollingworth, 2017-01-23:
I've just pushed a change to fix this to the 'next' branch (it may take some time for Dom to build a new version).  Please try again...

Thanks

Gordon

tvoverbeek, 2017-01-24:
@ghollingworth Now it works in a single run of usbboot. Loads kernel.img, bcm2708-rpi-b-plus.dtb and config.txt again. This is with commit f69368b from the next branch.
As far as I am concerned you can close the issue now.

closed 2017-01-24 by ghollingworth