raspberrypi/usbboot #16: MSD stopped being exposed

kubiko, 2017-04-19:
This has been working for me, but at some point CM3 goes to state where MSD is no more exposed on secondary USB port. As matter of fact there seems to be no USB device exposed at all.
usbboot does not show any obvious error and second stage bootloader is loaded to the device successfully.
lsusb on host computer does not show anything connected on USB though once second stage bootloader is loaded.
I have now two CM3 in this state. Filesystem on CM3 itself seems to be still fine, as normal boot still works, however I can't get it anymore to MSD mode.
I can't really say how to get CM3 to this state either, as all I was doing was debugging u-boot boot
Any thoughts?
Or may be adding boot artefacts which would have option to reset emmc on CM3?

ghollingworth, 2017-04-22:
Is this using the official CMIO carrier board?  It's possible that the eMMC is not being detected because the EMMC_DISABLE_N signal is not being correctly pulled high by the msd.elf code.

It may be the same issue as we're seeing on the Pi Zero where there is a bug in the bootcode.bin which in some cases doesn't terminate the communication correctly and as such gets stuck waiting for the boot.  Does the LED change state at all during this process?

It might be worth trying the usb_reset fix detailed elsewhere here since that would have a similar response to the one you're seeing

Gordon

kubiko, 2017-04-24:
yes this is using official Compute module 3 I/O carrier board.
Board starts with red + green LED on when I connect power, then green LED goes off somewhere after sending bootcode.bin. Green LED does not come back anymore.
I'm using u-boot on boot partition, if that helps to narrow ir. I thought that u-boot might have corrupted drive but erasing first few blocks on emmc did not help (only broke u-boot:) )
I just tested usb_reset merged 2 days ago and no joy, still same. No MSD is exposed.

ghollingworth, 2017-04-24:
Now you've erased the first couple of sectors can you try moving J4 from EN
to DIS.  This will mean that the eMMC will be enabled at boot time but
there's no MBR so it will fail the boot and drop back to USB boot...  Does
this make any difference?

Gordon

kubiko, 2017-04-24:
not much change.
I have actually two broken CM3, one on which I have erased few first sectors.
When I've moved J4 to DIS, CM3 with erased sectors now defaults to USB boot, but still no MSD once 'usbboot' completes.
other one will boot to u-boot, regardless, I guess that's expected as I moved J4….

ghollingworth, 2017-05-31:
I have updated the code to fix this problem... Please reopen if problems are still found.

closed 2017-05-31 by ghollingworth

ghollingworth, 2017-06-02:
Leave it at EN...

Check a couple of things:

1) Are you running rpiboot on a Pi, a PC or Mac?
2) Are you definite that you rebuilt rpiboot and are using sudo ./rpiboot
to make sure it picks up the new one?
3) What is the output on the command line when you run it
4) What does dmesg look like?

Gordon