raspberrypi/usbboot #20: never stops

Tychus, 2017-08-15:
i'm running this on a rpi3 to flash a CM3. running `./rpiboot` never exits after killing the process the device `/dev/sdX` is there and working.

running `./rpiboot -v -l` also shows that it can't find some file/files.

alicemirror, 2017-08-15:
I have done a series of systematic tests trying to understand this issue. I am almost certain that it is a software problem as I have done the tests on two different I/O boards and two different CM3 (as a matter of fact two complete kits). I am following these tests collecting also information on the reliability and potential issues/workarounds on the CM3 kit as I am road testing the device for Element14. The tests done today always reach the same point: the second phase find the Broadcom USB then read the first file and stop never ending with a list of missing files.

**Host platforms used:**

- Intel PC running Debian Jessie desktop
- Raspberry PI modle B3
- Compute module 3 board running the CM3 lite booting from a microSD as host and a second CM board running as slave.

Always same result. I am setting a Windows10 machine to test also this vase (not great hope frankly). I will post further updates tomorrow.

_[Note: Casually I have run the usbboot tool also in a case without the slave enable switch correctly set but the result was the same.]_

jviotti, 2017-08-16:
@alicemirror I've been recently running the software on Windows 10, and confirm it all seems to work fine.

@Tychus Can you show us the actual log?

Tychus, 2017-08-16:
on Windows it works fine almost all the time (i've had some issues where it won't detect the device or keep running forever when it did)
On linux it never worked ( at least not like they show in the [documentation](https://www.raspberrypi.org/documentation/hardware/computemodule/cm-emmc-flashing.md)).

```
sudo ./rpiboot
Waiting for BCM2835/6/7
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7
Second stage boot server
File read: start.elf
```

```
sudo ./rpiboot -v -l
Waiting for BCM2835/6/7
Device located successfully
Initialised device correctly
Found serial number 0
Sending bootcode.bin
libusb_bulk_transfer returned 0
Writing 50216 bytes
libusb_bulk_transfer returned 0
Successful read 4 bytes
Waiting for BCM2835/6/7
Device located successfully
Initialised device correctly
Found serial number 1
Second stage boot server
Received message GetFileSize: autoboot.txt
Cannot open file autoboot.txt
Received message GetFileSize: config.txt
Cannot open file config.txt
Received message GetFileSize: recovery.elf
Cannot open file recovery.elf
Received message GetFileSize: start.elf
File size = 433616 bytes
Received message ReadFile: start.elf
File read: start.elf
libusb_bulk_transfer returned 0
Received message GetFileSize: fixup.dat
Cannot open file fixup.dat
```

lurch, 2017-08-16:
I can confirm that (on Linux at least) after @ghollingworth 's recent changes, `sudo ./rpiboot -v` now successfully prints `Second stage boot server done` and cleanly exits, whereas before I always used to see the same behaviour as @Tychus where it would 'hang' after printing `Cannot open file fixup.dat`
:tada:

Tychus, 2017-08-16:
Works like a charm now. 
Thanks @ghollingworth

closed 2017-08-16 by Tychus

alicemirror, 2017-08-16:
Hi guys! I have finished just now testing the Windows 10 version and it works, but the issue remain, simply it is easy to by pass. When I launch the rpiboot under Windows 10 it starts a terminal and - exactly in the same way I experienced under Linux - the program starts and seems looping or waiting indefinitely after reading the first file at the second stage.

**What is the difference?** It is that despite this issue Windows recognises the CM3 as an external removable device, I format it then I copy the raspbian image. The rpiboot terminal can be ignore or closed forcely without any parceptual difference. Then the CM3 boots regularly.

Enrico

lurch, 2017-08-16:
Just a small point, but there's no need to format the CM3 before writing the Raspbian image to it :)

ghollingworth, 2017-08-16:
I have now updated both the Windows installer and the rpiboot code to fix this problem, please try again.  Also note I've now added the installer generation code to the usbboot github tree...

For latest version check out:

https://github.com/raspberrypi/usbboot/raw/master/win32/rpiboot_setup.exe

Thanks

Gordon