raspberrypi/usbboot #22: Unknown issue preventing flashing

sklv, 2017-08-28:
I am trying to flash a CM3 using an IO board. I have built the latest rpiboot:
```
$> sudo ./rpiboot
Waiting for BCM2835/6/7
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7
Second stage boot server
File read: start.elf
Second stage boot server done
```
```
$> lsblk | tail -n 3
sdg                    8:96   1   3.7G  0 disk
├─sdg1                 8:97   1  41.8M  0 part
└─sdg2                 8:98   1   3.6G  0 part
```
The issue is dd falsely claiming the transfer is working:
```
$> sudo dd if=base.img of=/dev/sdg bs=4MiB status=progress
1577058304 bytes (1.6 GB, 1.5 GiB) copied, 1.00206 s, 1.6 GB/s
442+1 records in
442+1 records out
1854418944 bytes (1.9 GB, 1.7 GiB) copied, 1.18432 s, 1.6 GB/s
```
in dmesg:
```
[52248.793936] usb 1-6: new high-speed USB device number 24 using ehci-pci
[52248.932326] usb-storage 1-6:1.0: USB Mass Storage device detected
[52248.932672] scsi host6: usb-storage 1-6:1.0
[52249.958182] scsi 6:0:0:0: Direct-Access     RPi-MSD- 0001                  PQ: 0 ANSI: 2
[52249.958828] sd 6:0:0:0: Attached scsi generic sg6 type 0
[52249.960029] sd 6:0:0:0: [sdg] 7634944 512-byte logical blocks: (3.91 GB/3.64 GiB)
[52249.960784] sd 6:0:0:0: [sdg] Write Protect is off
[52249.960788] sd 6:0:0:0: [sdg] Mode Sense: 0f 00 00 00
[52249.961597] sd 6:0:0:0: [sdg] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[52249.967401]  sdg: sdg1 sdg2
[52249.970137] sd 6:0:0:0: [sdg] Attached SCSI removable disk
```
Please let me know and I'll provide any necessary info.

pelwell, 2017-08-28:
Do you have the jumper set in the correct position? It looks to me like the CM3 is trying to boot instead of running the flashing image.

sklv, 2017-08-28:
J4 is set to EN which i believe is correct. When the board is powered up, ACT and PWR are on. When the 2nd ```Waiting for BCM2835/6/7``` is printed, the ACT led goes off but the PWR led stays on.

ghollingworth, 2017-09-26:
Did this go away or are you still having trouble?

What computer are you running the server on?  A Raspberry Pi or x86?  What distribution are you running?

Gordon

ghollingworth, 2018-04-06:
Can you try again, I've recently fixed a problem...  Use the latest usbboot c3ee230e commit...

closed 2018-04-06 by ghollingworth