raspberrypi/usbboot #24: RpiBoot problem

MartinSchulze1, 2017-10-10:
Hy,

I got an issue when i try to detect the drive letter after the rpiboot finishes.
I am using windows and the rpiboot.exe which comes with the installer.
(actually the sources here are not the same, these here are newer)

My motivation is to get more a unique identifier to enable
parallel flash multiple rpi3-compute module boards at the same time.
in windows, when rpiboot finishes, its always the last letter.

but the procedure gets confused sometimes, still to figure out for me...
its just because, rpiboot and msd-drive detection are separate.

Network would be an option (like mentioned in the previous issue) but i want to try usb only first.

My only clue for an unique-"id", which helps a lot, is the Hardware id of the drive, which is always Rpi-MSD-0001
It does not change.
It does change when i manually edid the "Rpi-MSD" suffix into something else with a hex editor
in msd.elf.

-linux probably works but not windows

is there something i can do?
I can patch the msd.elf to change the suffix 
(its in the file : )
own.Jungo...RPi.MSD.TASK.%s.R.W.on.off.on.off..

but i like to change the 0001 into 0002 and so on...
replacing the %s by 0001 does not help.

Is there a source of msd.elf?

Thanks for your help,

Martin

ghollingworth, 2017-10-10:
Theoretically you can create your own msd.elf by booting a linux kernel with gadget support and MSD support.  You'll also need to drive the GPIO47 suitably to re-enable the eMMC after booting though

Take a look at the https://github.com/raspberrypi/scriptexecutor for help with building your own bit of code to do this...

Other possibility is to enable the gadget mode with ethernet device support so the Raspberry Pi will appear on the PC as another network port

MartinSchulze1, 2018-01-09:
Thanks, i had a look into that. Actually i cirumvented the problem by registering to the wmi-event. its an event-handler, which notifies me with the usb-plug event.
That plug-event seems to be or is always the event from the previous execution of cmboot/rpiboot.exe

I did not know about scriptexecutor. 
Gadget mode under windows7 as a usb-client, i think probably under linux i am able to do that i guess..

problem is also solved by convention, means, plug one cm3, then wait till finished, then plug the next.

closed 2018-02-20 by ghollingworth