raspberrypi/usbboot #28: usbboot fails on a windows

MartinSchulze1, 2018-01-09:
Hy,

very precise error report above.
I cannot really explain the cause.

On some machines (windows version does not matter) rpiboot fails.
Here the output from "rpiboot -v"

Found device 4 idVendor=0x04f2 idProduct=0xb51c
Bus: 1, Device: 5 Path: 1-9
Found device 5 idVendor=0x0a5c idProduct=0x2764
Bus: 1, Device: 19 Path: 1-1.3
Device located successfully
libusb: error [parse_endpoint] invalid extra ep desc len (0)
libusb: error [raw_desc_to_config] parse_configuration failed with error -1
libusb: warning [winusbx_claim_interface] auto-claimed interface 0 (required to
claim 1 with WinUSB)
Failed to claim interface

-
In the above log-messages, the messages prefixed by "libusb:"
are showing that the endpoint cannot be parsed and thus cannot send bootcode.bin.

Ignoring the error (compile by vs-2012) does not help.

I could be a driver issue, but it could also be solved by the app.
I dont know, a direction would be appreciated.

(I am not a driver expert and looking into libusb is a bit ... on most machines i obeserved, it does work)

i have also a usb-tree good-bad diff...

From the usb-treeview on both machines, left side results in good cmboot, right side bad -.

[image]

Thanks, maybe this might be interesting because some machines dont work.
(i seen at least two)

Martin

ghollingworth, 2018-01-09:
What x86 hardware is this?  I think this is possibly a problem with the Windows PC hardware complaining about a response being too long.  This is a known issue and thought only to effect Macs, but it's possible it's not limited to the Mac

MartinSchulze1, 2018-01-09:
runnig dx-diag and display-related sections removed...
[DxDiag.txt](https://github.com/raspberrypi/usbboot/files/1616120/DxDiag.txt)

ghollingworth, 2018-02-20:
Not sure what it is that is wrong in this, I'd really need to run the USB analyser in the failing case to understand what is different...

Is this only on a specific Windows 7 computer?  Does it happen with all compute modules or just this one in particular...

Gordon

ghollingworth, 2018-04-06:
Closing assuming it is the same problem as seen on a small number of Mac machines (assuming they share the same USB root hub hardware.

closed 2018-04-06 by ghollingworth