raspberrypi/usbboot #32: make: *** [rpiboot] Error 1

samuelbles07, 2018-05-24:
Hi,

First of all i follow this [link](http://www.uugear.com/portfolio/flashing-the-raspberry-pi-compute-module-emmc-under-mac-os/) for installation.

When i install usbboot in my mac, i got this error:
~~~
cc -Wall -Wextra -g -o bin2c bin2c.c
./bin2c msd/bootcode.bin msd/bootcode.h
./bin2c msd/start.elf msd/start.h
cc -Wall -Wextra -g -o rpiboot main.c -lusb-1.0
main.c:394:16: warning: comparison of array 'pathname' not equal to a null pointer is always
      true [-Wtautological-pointer-compare]
                if(overlay&&(pathname != NULL))
                             ^~~~~~~~    ~~~~
main.c:417:9: warning: implicit declaration of function 'fmemopen' is invalid in C99
      [-Wimplicit-function-declaration]
                        fp = fmemopen(msd_bootcode_bin, msd_bootcode_bin_len, "r");
                             ^
main.c:417:7: warning: incompatible integer to pointer conversion assigning to 'FILE *'
      (aka 'struct __sFILE *') from 'int' [-Wint-conversion]
                        fp = fmemopen(msd_bootcode_bin, msd_bootcode_bin_len, "r");
                           ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
main.c:420:8: warning: incompatible integer to pointer conversion assigning to 'FILE *'
      (aka 'struct __sFILE *') from 'int' [-Wint-conversion]
                                fp = fmemopen(msd_start_elf, msd_start_elf_len, "r");
                                   ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
main.c:601:2: warning: 'libusb_set_debug' is deprecated [-Wdeprecated-declarations]
        libusb_set_debug(ctx, verbose ? LIBUSB_LOG_LEVEL_WARNING : 0);
        ^
/usr/local/include/libusb-1.0/libusb.h:1299:1: note: 'libusb_set_debug' has been explicitly
      marked deprecated here
LIBUSB_DEPRECATED_FOR(libusb_set_option)
^
/usr/local/include/libusb-1.0/libusb.h:89:49: note: expanded from macro 'LIBUSB_DEPRECATED_FOR'
#define LIBUSB_DEPRECATED_FOR(f) __attribute__((deprecated))
                                                ^
5 warnings generated.
Undefined symbols for architecture x86_64:
  "_fmemopen", referenced from:
      _check_file in main-6a807a.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [rpiboot] Error 1
~~~

I already install lib usb that i directly download from [here](https://sourceforge.net/projects/libusb/?source=typ_redirect)

When i do `make` here what i got 

~~~
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in libusb
  CC       libusb_1_0_la-core.lo
  CC       libusb_1_0_la-descriptor.lo
  CC       libusb_1_0_la-hotplug.lo
  CC       libusb_1_0_la-io.lo
  CC       libusb_1_0_la-strerror.lo
  CC       libusb_1_0_la-sync.lo
  CC       os/libusb_1_0_la-poll_posix.lo
  CC       os/libusb_1_0_la-threads_posix.lo
  CC       os/libusb_1_0_la-darwin_usb.lo
  CCLD     libusb-1.0.la
Making all in doc
make[2]: Nothing to be done for `all'.
make[2]: Nothing to be done for `all-am'.
~~~

Then install 

~~~
Making install in libusb
 .././install-sh -c -d '/usr/local/lib'
 /bin/sh ../libtool   --mode=install /usr/bin/install -c   libusb-1.0.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libusb-1.0.0.dylib /usr/local/lib/libusb-1.0.0.dylib
libtool: install: (cd /usr/local/lib && { ln -s -f libusb-1.0.0.dylib libusb-1.0.dylib || { rm -f libusb-1.0.dylib && ln -s libusb-1.0.0.dylib libusb-1.0.dylib; }; })
libtool: install: /usr/bin/install -c .libs/libusb-1.0.lai /usr/local/lib/libusb-1.0.la
libtool: install: /usr/bin/install -c .libs/libusb-1.0.a /usr/local/lib/libusb-1.0.a
libtool: install: chmod 644 /usr/local/lib/libusb-1.0.a
libtool: install: ranlib /usr/local/lib/libusb-1.0.a
 .././install-sh -c -d '/usr/local/include/libusb-1.0'
 /usr/bin/install -c -m 644 libusb.h '/usr/local/include/libusb-1.0'
Making install in doc
make[2]: Nothing to be done for `install-exec-am'.
make[2]: Nothing to be done for `install-data-am'.
make[2]: Nothing to be done for `install-exec-am'.
 ./install-sh -c -d '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 libusb-1.0.pc '/usr/local/lib/pkgconfig'
~~~

I need rpiboot for flashing compute module for my project. Is there something that i miss?

ghollingworth, 2018-05-24:
It looks like Mac OS is missing the fmemopen function, have you got the latest OS?  From random googling it looks like it may have been added at Mac OS 10.5

samuelbles07, 2018-05-24:
nope, i still using the sierra 10.12.6

ghollingworth, 2018-05-24:
OK,

In that case you'll have to update your operating system or use one of the open source versions of fmemopen

Gordon

closed 2018-05-24 by ghollingworth