raspberrypi/usbboot #33: Newest commits do not work on Debian Jessie

Tilogorn, 2018-05-30:
As we pulled and make'd the latest commits yesterday on out up-to-date Debian Jessie PC, suddenly CM3-Modules couldn't be mounted anymore. After upgrading to Debian Stretch, everything worked fine again.

`sudo ./rpiboot` tried to send the bootcode three times and failed in an error at third time (smth. like "Answer was too short") after about a minute. Unfortunately we didn't save the output before upgrading to Stretch.

Last working commit on Debian Jessie (Apr 5, 2018):
c3ee230ef4a8f247b7b7d406e59beec8c895de60

First commit where error occurs (Apr 12, 2018)
1bb4c2da4775329bcf5811616fc8f0ce78ee2121

If it is still interesting to support Debian Jessie, it should be easily reproducable on a machine with Debian Jessie and current updates.

ghollingworth, 2018-05-30:
I would guess the problem is between the two different versions of
libusb-1.0-0-dev (1.0.19-1 to 1.0.21-1)

closed 2022-02-23 by timg236

Andy1978, 2022-02-23:
Hurray, only 4 years of ignoring a problem and then saying: "it's too old, closing". This is not what I was expecting treating CM3 customers

timg236, 2022-02-23:
I can ignore it for longer

JamesH65, 2022-02-24:
Debian Jesse went out of LTS in June 2020, so if you are still using it you should upgrade as it is certainly a security risk as it would have had not updates in nearly two years. Even Stretch, which followed Jesse, goes out of LTS this year, so upgrading to a much more recent version would be strongly advised.

https://wiki.debian.org/LTS

Andy1978, 2022-07-14:
@JamesH65 I don't know if you've missed, but the original issue was submitted in May 2018, when Jessie was oldstable

JamesH65, 2022-07-14:
Jesse is no longer supported, you should really be upgrading to a much more recent system, which will have much better security. And will be much better overall.

Andy1978, 2022-07-15:
I think your last post has it all: It shows the complete ignorance and the lack of ability to read and understand text