raspberrypi/usbboot #35: Stuck waiting for BCM2835/6/7 on Mac

ball-hayden, 2018-06-21:
Environment:
MacOS High Sierra (10.13.3)
libusb stable 1.0.22 (installed via homebrew)
CM3 with CM3IO board (USB Slave Boot Enable is EN).

`make` succeeds, but displays deprecation warnings:

```
$ make
cc -Wall -Wextra -g -o bin2c bin2c.c
./bin2c msd/bootcode.bin msd/bootcode.h
./bin2c msd/start.elf msd/start.h
cc -Wall -Wextra -g -o rpiboot main.c -lusb-1.0
main.c:394:16: warning: comparison of array 'pathname' not equal to a null pointer is always true [-Wtautological-pointer-compare]
                if(overlay&&(pathname != NULL))
                             ^~~~~~~~    ~~~~
main.c:601:2: warning: 'libusb_set_debug' is deprecated [-Wdeprecated-declarations]
        libusb_set_debug(ctx, verbose ? LIBUSB_LOG_LEVEL_WARNING : 0);
        ^
/usr/local/include/libusb-1.0/libusb.h:1299:1: note: 'libusb_set_debug' has been explicitly marked deprecated here
LIBUSB_DEPRECATED_FOR(libusb_set_option)
^
/usr/local/include/libusb-1.0/libusb.h:89:49: note: expanded from macro 'LIBUSB_DEPRECATED_FOR'
#define LIBUSB_DEPRECATED_FOR(f) __attribute__((deprecated))
                                                ^
2 warnings generated.
```

Running `sudo ./rpiboot` appears to run as expected, but the pi is never detected - I see "Waiting for BCM2835/6/7" forever.

There doesn't appear to be anything in the syslog to show the pi appearing as a USB device.

ball-hayden, 2018-06-21:
Thanks for your prompt response.
This is using a CM3, so it is the 2837...

burtyb, 2018-06-21:
To fix the first compiler warning you probably want https://github.com/raspberrypi/usbboot/pull/34/commits/838b8753b848e86015b81f106e9c69a9035a9fd6

ball-hayden, 2018-06-22:
Thanks @burtyb. I checked out https://github.com/raspberrypi/usbboot/pull/34 and tried again - still no luck unfortunately.
Is there anywhere you could suggest looking so I can provide more information to help debug this?
I'm not particularly familiar with the Mac platform myself, so maybe there's a different log I should be looking at?

ball-hayden, 2018-06-22:
Turns out it was a bad USB cable. Sorry for the noise 😦

closed 2018-06-22 by ball-hayden