raspberrypi/usbboot #41: Any Android ports?

DimaK-tracxpoint, 2019-07-04:
Did anybody try to port this service to Android?

cleverca22, 2020-07-24:
a quick glance at https://developer.android.com/guide/topics/connectivity/usb/host implies that it is likely possible, and could be a simple task, for somebody familiar with the code

cleverca22, 2020-10-20:
well, i solved this issue by accident, from the totally wrong direction
https://github.com/librerpi/rpi-tools/tree/master/webusbboot
this code uses webusb to flash an rpi4 (or cm4) over usb

i have successfully flashed my rpi4 eeprom from desktop chromium using this code
and if i power the rpi4 from the gpio header, and use an OTG adapter cable, i can run the exact same JS on my phone, and shove `msd/start4.elf` over, putting it into MSD mode

timg236, 2022-02-23:
No plans for an official Android port so closing.