raspberrypi/usbboot #43: Booting multiple RPis in Parallel

eli-tracxpoint, 2019-07-30:
Hi guys!
I'm using a non-SD card setup of 6 x RPi zero that booting with rpiboot and fetching the filesystem from NFS folder. 
The issue here is that booting them serially with rpiboot takes a lot of time.
Is there a way using rpiboot on multiple RPis in parallel instead of using the loop flag (-l) that boots serially.

Thanks!

ghollingworth, 2019-07-30:
You'd need to rewrite rpiboot to run a new thread for each device as it enumerates, then each thread runs the same bit of code

lurch, 2019-11-07:
I wonder if https://github.com/balena-io-modules/node-raspberrypi-usbboot might have better threading support? (although I've no idea if it'll be possible to adapt it to support your scenario)

cleverca22, 2020-10-29:
https://github.com/librerpi/rpi-tools/tree/master/webusbboot

another issue ive found a potential solution to by accident
webusb is entirely event driven, so one thread can manage multiple devices at once

webusbboot has been confirmed to be able to:
* reflash the rpi4b eeprom (both with `recovery.bin`, and eeprom self-reflash)
* boot into VPU MSD mode (with `msd/start4.elf`)
* boot into linux MSD mode (tested with [nixos-msd](https://github.com/librerpi/rpi-tools/tree/master/nixos-msd) but others should work too)

it works on linux (even android), windows has general webusb issues, darwin untested

burtyb, 2020-10-29:
With https://github.com/raspberrypi/usbboot/pull/66 you could also run rpiboot for each Pi you want to boot or use the existing -1/-2/etc. option if the port numbers aren't repeated (as they are on the 3B+).

timg236, 2024-01-08:
Closed due to inactivity.

closed 2024-01-08 by timg236