raspberrypi/usbboot #44: requiring sudo is just plain wrong

laf0rge, 2019-08-29:
A program that at runtime checks if it's running as root, rather than checking for the *actual* permissions that it requires.  I don't think I've seen such bad practise any time before, to be honest.

I don't really see why the program would require to run as root.  Root privileges are *way* too powerful.  All that's needed appears to be access to some USB devices.  So the proper solution is to ship udev rules for those vid/pid combinations, rather than forcing people to run code with way too high privileges.

ghollingworth, 2019-08-29:
Sure, would be happy to look at the pull request

QuLogic, 2019-09-11:
Isn't that #27?

lurch, 2020-10-21:
I guess this can be closed, now that #27 has been merged?

closed 2020-10-21 by ghollingworth