raspberrypi/usbboot #47: Compilation error in fmemopen.c

petronny, 2020-02-05:
Getting when building [rpiusbboot-git](https://aur.archlinux.org/packages/rpiusbboot-git) on latest archlinux.
```
In file included from main.c:13:
fmemopen.c: In function ‘seekfn’:
fmemopen.c:62:18: error: invalid operands to binary >= (have ‘fpos_t’ {aka ‘struct _G_fpos_t’} and ‘int’)
   62 |       if (offset >= 0) {
      |                  ^~
fmemopen.c:63:9: error: aggregate value used where an integer was expected
   63 |         pos = (size_t)offset;
      |         ^~~
fmemopen.c:70:18: error: invalid operands to binary >= (have ‘fpos_t’ {aka ‘struct _G_fpos_t’} and ‘int’)
   70 |       if (offset >= 0 || (size_t)(-offset) <= mem->pos) {
      |                  ^~
fmemopen.c:70:35: error: wrong type argument to unary minus
   70 |       if (offset >= 0 || (size_t)(-offset) <= mem->pos) {
      |                                   ^
fmemopen.c:71:9: error: aggregate value used where an integer was expected
   71 |         pos = mem->pos + (size_t)offset;
      |         ^~~
fmemopen.c:77:5: error: aggregate value used where an integer was expected
   77 |     case SEEK_END: pos = mem->size + (size_t)offset; break;
      |     ^~~~
fmemopen.c:78:21: error: incompatible types when returning type ‘int’ but ‘fpos_t’ {aka ‘struct _G_fpos_t’} was expected
   78 |     default: return -1;
      |                     ^
fmemopen.c:82:12: error: incompatible types when returning type ‘int’ but ‘fpos_t’ {aka ‘struct _G_fpos_t’} was expected
   82 |     return -1;
      |            ^
fmemopen.c:86:3: error: conversion to non-scalar type requested
   86 |   return (fpos_t)pos;
      |   ^~~~~~
```
Full log:
```
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
:: Starting full system upgrade...
 there is nothing to do
==> Building in chroot for [extra] (x86_64)...
==> Synchronizing chroot copy [/var/lib/archbuild/extra-x86_64/root] -> [calil]...done
==> Making package: rpiusbboot-git r73.45c7237-1 (Wed Feb  5 01:07:21 2020)
==> Retrieving sources...
  -> Updating usbboot git repo...
Fetching origin
==> Validating source files with md5sums...
    usbboot ... Skipped
==> Making package: rpiusbboot-git r73.45c7237-1 (Wed 05 Feb 2020 01:07:30 AM CST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Installing missing dependencies...
resolving dependencies...
looking for conflicting packages...

Packages (4) perl-error-0.17029-1  perl-mailtools-2.21-2  perl-timedate-2.31-1
             git-2.25.0-1

Total Installed Size:  37.65 MiB

:: Proceed with installation? [Y/n] 
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
:: Processing package changes...
installing perl-error...
installing perl-timedate...
installing perl-mailtools...
installing git...
Optional dependencies for git
    tk: gitk and git gui
    perl-libwww: git svn
    perl-term-readkey: git svn and interactive.singlekey setting
    perl-mime-tools: git send-email
    perl-net-smtp-ssl: git send-email TLS support
    perl-authen-sasl: git send-email TLS support
    perl-mediawiki-api: git mediawiki support
    perl-datetime-format-iso8601: git mediawiki support
    perl-lwp-protocol-https: git mediawiki https support
    perl-cgi: gitweb (web interface) support
    python: git svn & git p4
    subversion: git svn
    org.freedesktop.secrets: keyring credential helper
    libsecret: libsecret credential helper [installed]
:: Running post-transaction hooks...
(1/4) Creating system user accounts...
Creating group git with gid 977.
Creating user git (git daemon user) with uid 977 and gid 977.
(2/4) Reloading system manager configuration...
  Skipped: Current root is not booted.
(3/4) Arming ConditionNeedsUpdate...
(4/4) Warn about old perl modules
==> Retrieving sources...
==> WARNING: Skipping all source file integrity checks.
==> Extracting sources...
  -> Creating working copy of usbboot git repo...
Cloning into 'usbboot'...
done.
==> Starting pkgver()...
==> Starting build()...
cc -Wall -Wextra -g -o bin2c bin2c.c
./bin2c msd/bootcode.bin msd/bootcode.h
./bin2c msd/start.elf msd/start.h
cc -Wall -Wextra -g -o rpiboot main.c -lusb-1.0
In file included from main.c:13:
fmemopen.c: In function ‘seekfn’:
fmemopen.c:62:18: error: invalid operands to binary >= (have ‘fpos_t’ {aka ‘struct _G_fpos_t’} and ‘int’)
   62 |       if (offset >= 0) {
      |                  ^~
fmemopen.c:63:9: error: aggregate value used where an integer was expected
   63 |         pos = (size_t)offset;
      |         ^~~
fmemopen.c:70:18: error: invalid operands to binary >= (have ‘fpos_t’ {aka ‘struct _G_fpos_t’} and ‘int’)
   70 |       if (offset >= 0 || (size_t)(-offset) <= mem->pos) {
      |                  ^~
fmemopen.c:70:35: error: wrong type argument to unary minus
   70 |       if (offset >= 0 || (size_t)(-offset) <= mem->pos) {
      |                                   ^
fmemopen.c:71:9: error: aggregate value used where an integer was expected
   71 |         pos = mem->pos + (size_t)offset;
      |         ^~~
fmemopen.c:77:5: error: aggregate value used where an integer was expected
   77 |     case SEEK_END: pos = mem->size + (size_t)offset; break;
      |     ^~~~
fmemopen.c:78:21: error: incompatible types when returning type ‘int’ but ‘fpos_t’ {aka ‘struct _G_fpos_t’} was expected
   78 |     default: return -1;
      |                     ^
fmemopen.c:82:12: error: incompatible types when returning type ‘int’ but ‘fpos_t’ {aka ‘struct _G_fpos_t’} was expected
   82 |     return -1;
      |            ^
fmemopen.c:86:3: error: conversion to non-scalar type requested
   86 |   return (fpos_t)pos;
      |   ^~~~~~
fmemopen.c: In function ‘fmemopen’:
fmemopen.c:95: warning: ignoring #pragma unused  [-Wunknown-pragmas]
   95 |   #pragma unused(mode)
      | 
fmemopen.c:106:10: warning: implicit declaration of function ‘funopen’; did you mean ‘fdopen’? [-Wimplicit-function-declaration]
  106 |   return funopen(mem, readfn, writefn, seekfn, closefn);
      |          ^~~~~~~
      |          fdopen
fmemopen.c:106:10: warning: returning ‘int’ from a function with return type ‘FILE *’ {aka ‘struct _IO_FILE *’} makes pointer from integer without a cast [-Wint-conversion]
  106 |   return funopen(mem, readfn, writefn, seekfn, closefn);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
fmemopen.c:94:52: warning: unused parameter ‘mode’ [-Wunused-parameter]
   94 | FILE *fmemopen(void *buf, size_t size, const char *mode) {
      |                                        ~~~~~~~~~~~~^~~~
main.c: In function ‘main’:
main.c:620:2: warning: ‘libusb_set_debug’ is deprecated: Use libusb_set_option instead [-Wdeprecated-declarations]
  620 |  libusb_set_debug(ctx, verbose ? LIBUSB_LOG_LEVEL_WARNING : 0);
      |  ^~~~~~~~~~~~~~~~
In file included from main.c:1:
/usr/include/libusb-1.0/libusb.h:1325:18: note: declared here
 1325 | void LIBUSB_CALL libusb_set_debug(libusb_context *ctx, int level);
      |                  ^~~~~~~~~~~~~~~~
In file included from main.c:13:
fmemopen.c: In function ‘seekfn’:
fmemopen.c:87:1: warning: control reaches end of non-void function [-Wreturn-type]
   87 | }
      | ^
make: *** [Makefile:2: rpiboot] Error 1
==> ERROR: A failure occurred in build().
    Aborting...
```

ghollingworth, 2020-02-05:
@jrmithdobbs 

Can you look into this, or we'll have to revert the commit

pelwell, 2020-02-05:
https://github.com/raspberrypi/usbboot/commit/17f6b01ef5ff545f58db11d98d020e8f3c977dd7  should fix it for most cases.

petronny, 2020-02-06:
Problem solved. Thanks!

closed 2020-02-06 by petronny