raspberrypi/usbboot #51: VisualStudio Code Error in Code

TRiXXER76, 2020-03-10:
Hello,

I would like to Build (compile) the Programm but I get an error.

[image]

The Error Message says "The Identifier in ""msd_start_elf"" is not defined."
the same Error with msd_bootcode_bin, msd_bootcode_bin_len and msd_start_elf_len. Can maybe someone help me ? 

Greetings Lukas

pelwell, 2020-03-10:
If you look in the Makefile you'll see that rpiboot depends on rpiboot.c, msd/bootcode.h and msd/start.h. Those files don't exist in the repo. but the Makefile knows how to build a .h file from a .bin file or a .elf file using bin2c, and also how to build bin2c from source.

If you're not that familiar with the VisualStudio build system you could build bin2c as a separate step and use it to generate the .h files:
```
$ cd msd
$ ../bin2c bootcode.bin bootcode.h
$ ../bin2c start.elf start.h
```

TRiXXER76, 2020-03-10:
Okay thanks. That makes everything a bit more clear to me. I added the bin2c.exe to my Environment Variables but now I get this Error in cmd:

> C:\Users\lkn\Desktop\CodeProjects\usbboot\msd>bin2c bootcode.bin bootcode.h
> bin2c: No output files are named

pelwell, 2020-03-10:
I think you've picked up the wrong bin2c. Use the relative (or absolute, if you prefer) path as I did in my example.

pelwell, 2020-03-10:
Yes - that's some other bin2c, but if it works then great.

TRiXXER76, 2020-03-10:
Ok now the Problem with the Header files is sorted out... but ist still not compiling. It still says the msd_bootcode_bin etc. are not declared :/

pelwell, 2020-03-10:
I'm sure that's because you are using the wrong bin2c. The output produced by the version in the repo begins:
```
/* Automatically generated file from msd/bootcode.bin */
unsigned int msd_bootcode_bin_len = 51992;
unsigned char msd_bootcode_bin[] = {
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
```
whereas I suspect your output starts with:
```
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
```
Use the version provided and all will be well.

TRiXXER76, 2020-03-10:
Okay I will try.

/* Generated by bin2c, do not edit manually */

```
/* Contents of file bootcode.bin */
const long int bootcode_bin_size = 51992;
const unsigned char bootcode_bin[51992] = {
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
```

Thats the content of my bootcode.h

pelwell, 2020-03-10:
I'm sure that will work.

TRiXXER76, 2020-03-10:
Seems like it will not work so easy... I compiled the bin2c.c file to a bin2c.exe. When I try to execute your command it says that the cygwin1.dll is missing. So I copied my cygwin1.dll and the 2 others in the directory but then I get an error that it cant be executed.

The Error is:

[image]

TRiXXER76, 2020-03-10:
Translation: The Application could not be started correctly ( 0xc000007b). Click "OK" to close the Application.

pelwell, 2020-03-10:
If you are using gcc (likely, with cygwin) then you can rebuild with:
```
$ gcc -static -o bin2c bin2c.c
```

TRiXXER76, 2020-03-10:
I just generated the header files using the make script on a Linux Subsystem. Now the Errors in VC Code are gone but it still cant compile the main.c

```
> Executing task: E:/cygwin64/bin/gcc.exe -g c:\Users\lkn\Desktop\CodeProjects\usbboot\main.c -o c:\Users\lkn\Desktop\CodeProjects\usbboot\main.exe <

c:\Users\lkn\Desktop\CodeProjects\usbboot\main.c: In Funktion »main«:
c:\Users\lkn\Desktop\CodeProjects\usbboot\main.c:621:2: Warnung: »libusb_set_debug« ist veraltet: Use libusb_set_option instead [-Wdeprecated-declarations]
  libusb_set_debug(ctx, verbose ? LIBUSB_LOG_LEVEL_WARNING : 0);
  ^~~~~~~~~~~~~~~~
In file included from c:\Users\lkn\Desktop\CodeProjects\usbboot\main.c:1:0:
c:\Users\lkn\Desktop\CodeProjects\usbboot\libusb.h:1325:18: Anmerkung: hier deklariert
 void LIBUSB_CALL libusb_set_debug(libusb_context *ctx, int level);
                  ^~~~~~~~~~~~~~~~
/cygdrive/c/Users/lkn/AppData/Local/Temp/ccbUeqHf.o: In Funktion »open_device_with_vid«:
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:71: Warnung: undefinierter Verweis auf »libusb_get_device_list«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:71:(.text+0x283): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_get_device_list'
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:77: Warnung: undefinierter Verweis auf »libusb_get_device_descriptor«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:77:(.text+0x2ad): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_get_device_descriptor'
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:83: Warnung: undefinierter Verweis auf »libusb_get_port_numbers«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:83:(.text+0x2e9): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_get_port_numbers'
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:84: Warnung: undefinierter Verweis auf »libusb_get_bus_number«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:84:(.text+0x2f8): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_get_bus_number'
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:101: Warnung: undefinierter Verweis auf »libusb_get_port_number«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:101:(.text+0x3f3): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_get_port_number'
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:106: Warnung: undefinierter Verweis auf »libusb_get_device_address«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:106:(.text+0x432): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_get_device_address'
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:106: Warnung: undefinierter Verweis auf »libusb_get_bus_number«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:106:(.text+0x441): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_get_bus_number'
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:137: Warnung: undefinierter Verweis auf »libusb_open«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:137:(.text+0x536): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_open'
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:151: Warnung: undefinierter Verweis auf »libusb_free_device_list«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:151:(.text+0x590): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_free_device_list'
/cygdrive/c/Users/lkn/AppData/Local/Temp/ccbUeqHf.o: In Funktion »Initialize_Device«:
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:169: Warnung: undefinierter Verweis auf »libusb_get_device«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:169:(.text+0x602): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `libusb_get_device'
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:169: Warnung: undefinierter Verweis auf »libusb_get_active_config_descriptor«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:169:(.text+0x611): additional relocation overflows omitted from the output
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:191: Warnung: undefinierter Verweis auf »libusb_claim_interface«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:194: Warnung: undefinierter Verweis auf »libusb_close«
/cygdrive/c/Users/lkn/AppData/Local/Temp/ccbUeqHf.o: In Funktion »ep_write«:
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:210: Warnung: undefinierter Verweis auf »libusb_control_transfer«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:223: Warnung: undefinierter Verweis auf »libusb_bulk_transfer«
/cygdrive/c/Users/lkn/AppData/Local/Temp/ccbUeqHf.o: In Funktion »ep_read«:
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:238: Warnung: undefinierter Verweis auf »libusb_control_transfer«
/cygdrive/c/Users/lkn/AppData/Local/Temp/ccbUeqHf.o: In Funktion »file_server«:
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:500: Warnung: undefinierter Verweis auf »libusb_control_transfer«
/cygdrive/c/Users/lkn/AppData/Local/Temp/ccbUeqHf.o: In Funktion »main«:
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:614: Warnung: undefinierter Verweis auf »libusb_init«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:621: Warnung: undefinierter Verweis auf »libusb_set_debug«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:635: Warnung: undefinierter Verweis auf »libusb_get_device«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:635: Warnung: undefinierter Verweis auf »libusb_get_device_descriptor«
[... cut ...]
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:647: Warnung: undefinierter Verweis auf »libusb_get_device«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:647: Warnung: undefinierter Verweis auf »libusb_get_active_config_descriptor«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:669: Warnung: undefinierter Verweis auf »libusb_close«
c:\Users\lkn\Desktop\CodeProjects\usbboot/main.c:674: Warnung: undefinierter Verweis auf »libusb_exit«
collect2: Fehler: ld gab 1 als Ende-Status zurück
The terminal process terminated with exit code: 1
```

TRiXXER76, 2020-03-10:
Can I get the Console output in VC Code in English ? Think thats better for you tho :D

pelwell, 2020-03-10:
These are linker errors - you need to include the usb-1.0 library in the linking stage. For gcc this means adding "-lusb-1.0" to the command line.

TRiXXER76, 2020-03-11:
Yeah I tried that... The Problem is in Visual Studio Code the Arguments seem to be different... -lusb-1.0 doesnt work, -lusb, -lusb[-1.0] and linking the library with -l libusb.a doesnt work too... Maybe someone knows how to link libusb with Visual Studio Code ?

TRiXXER76, 2020-03-11:
```
Executing task: C:\winbuilds\bin\gcc.exe -Wall -Wextra -g c:\Users\lkn\Desktop\CodeProjects\usbboot\.vscode\tasks.json -o c:\Users\lkn\Desktop\CodeProjects\usbboot\.vscode\tasks.exe -lusb-1.0 <

gcc.exe: error: .0: No such file or directory
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

Why is he cutting of the .0 from -lusb-1.0 ?

timg236, 2022-03-31:
Closing this old issue. We are only going to support gcc. If there's a real bug in the source code then feel free to submit a PR

closed 2022-03-31 by timg236