raspberrypi/usbboot #52: Circular loop between raspberrypi/usbboot title and raspberrypi/tools/usbboot readme.md

stevenhoving, 2020-04-05:
The git repo title says:
> Raspberry Pi USB booting code, moved from tools repository

The raspberrypi tools repository readme.md
https://github.com/raspberrypi/tools/blob/master/usbboot/Readme.md
Says:
> Please note, this code has now moved to
https://github.com/raspberrypi/usbboot.git

While I love these circular loops jokes, I can imagine that this was not done on purpose. Maybe fix this project title.

pelwell, 2020-04-05:
Huh? The tools repo says it has been moved to usbboot, while the usbboot repo says it has been moved from the tools repo.

stevenhoving, 2020-04-05:
> Huh? The tools repo says it has been moved to usbboot, while the usbboot repo says it has been moved from the tools repo.

yup

pelwell, 2020-04-05:
And that's a problem?

robindegen, 2020-04-05:
@pelwell Which one is correct? One of them has to be updated right?

stevenhoving, 2020-04-05:
not really, but just say that it can be confusing. I got a good laugh out of it when I noticed it.

stevenhoving, 2020-04-05:
> And that's a problem?

not really, but just say that it can be confusing. I got a good laugh out of it when I noticed it

pelwell, 2020-04-05:
It's not so much a circularity, more of a doubly-linked list. You could argue that the "from" comment is unnecessary, but it's a way of saying "this is the new home of the usbboot utility that you may have seen elsewhere."

stevenhoving, 2020-04-05:
Ah, now i see. I read over the 'from'' part. Sorry to cause confusion, especially on a Sunday.

closed 2020-04-05 by stevenhoving