raspberrypi/usbboot #53: main.c:620:2: warning: 'libusb_set_debug' is deprecated

geerlingguy, 2020-04-27:
Running this with the latest version of `libusb` on my Mac, I got the following warning (no error, through) when running `make`:

```
$ make          
cc -Wall -Wextra -g -o bin2c bin2c.c
./bin2c msd/bootcode.bin msd/bootcode.h
./bin2c msd/start.elf msd/start.h
cc -Wall -Wextra -g -o rpiboot main.c -lusb-1.0
main.c:620:2: warning: 'libusb_set_debug' is deprecated [-Wdeprecated-declarations]
        libusb_set_debug(ctx, verbose ? LIBUSB_LOG_LEVEL_WARNING : 0);
        ^
/usr/local/include/libusb-1.0/libusb.h:1324:1: note: 'libusb_set_debug' has been explicitly marked deprecated here
LIBUSB_DEPRECATED_FOR(libusb_set_option)
^
/usr/local/include/libusb-1.0/libusb.h:89:49: note: expanded from macro 'LIBUSB_DEPRECATED_FOR'
#define LIBUSB_DEPRECATED_FOR(f) __attribute__((deprecated))
                                                ^
1 warning generated.
```

necrifede, 2020-05-10:
same error on Linux:
```
user@host:~/usbboot$ make
cc -Wall -Wextra -g -o bin2c bin2c.c
./bin2c msd/bootcode.bin msd/bootcode.h
./bin2c msd/start.elf msd/start.h
cc -Wall -Wextra -g -o rpiboot main.c -lusb-1.0
main.c: In function ‘main’:
main.c:620:2: warning: ‘libusb_set_debug’ is deprecated: Use libusb_set_option instead [-Wdeprecated-declarations]
  libusb_set_debug(ctx, verbose ? LIBUSB_LOG_LEVEL_WARNING : 0);
  ^~~~~~~~~~~~~~~~
In file included from main.c:1:
/usr/include/libusb-1.0/libusb.h:1300:18: note: declared here
 void LIBUSB_CALL libusb_set_debug(libusb_context *ctx, int level);
                  ^~~~~~~~~~~~~~~~

```

My OS:
```
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
```

ghollingworth, 2020-05-10:
PRs welcome

closed 2020-05-10 by ghollingworth

geerlingguy, 2020-05-10:
Was it fixed or is it still open for a PR?

pelwell, 2020-05-10:
I'd say the timing makes that fairly obvious - the code has not been changed.

necrifede, 2020-05-10:
Hi @geerlingguy could you verify it please?

geerlingguy, 2020-05-11:
I opened https://github.com/raspberrypi/usbboot/pull/57

pelwell, 2020-05-11:
Did you notice #55?

geerlingguy, 2020-05-11:
@pelwell No, didn't until just now. Whatever fixes it is great for me; I'm just used to following an issue and wanted to make sure to get something in place before I forgot about it since this issue is closed :)

pelwell, 2020-05-11:
I've never had to choose between two competing PRs before! If you can see something wrong with #55 then please say so, otherwise it is likely to get merged once @ghollingworth has a moment.

geerlingguy, 2020-05-11:
@pelwell it also resolves the deprecation warning, and I don't have any particular opinion on the verbosity levels (I just hate seeing warnings in my terminal ;)—so please merge #55 as long as everything else is okay, and close #56.