raspberrypi/usbboot #69: Unable to usb boot raspberry pi zero v1.3

adrian-amaglio, 2020-11-17:
I got an error when I try to boot a pi zero from usb.
I am following this tutorial:
https://dev.webonomic.nl/how-to-run-or-boot-raspbian-on-a-raspberry-pi-zero-without-an-sd-card/comment-page-1

Since I do not understand what it is happening, I don’t know which information you need to understand the issue.

I start `usbboot`, then I plug the pi zero. All following logs are printed, then nothing happens anymore.
Execution trace
```
$  sudo ./rpiboot -v -d boot/
Boot directory 'boot/'
Loading: boot//bootcode.bin
Waiting for BCM2835/6/7/2711...
Loading: boot//bootcode.bin
Device located successfully
libusb: error [udev_hotplug_event] ignoring udev action bind
Initialised device correctly
Found serial number 0
Sending bootcode.bin
libusb_bulk_transfer sent 24 bytes; returned 0
Writing 52480 bytes
libusb_bulk_transfer sent 52480 bytes; returned 0
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
^C
```
In the same time, dmesg logged :
```
[ 2103.768199] usb 1-2: new full-speed USB device number 18 using xhci_hcd
[ 2103.909147] usb 1-2: New USB device found, idVendor=0a5c, idProduct=2763, bcdDevice= 0.00
[ 2103.909153] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2103.909155] usb 1-2: Product: BCM2708 Boot
[ 2103.909158] usb 1-2: Manufacturer: Broadcom
[ 2106.556630] usb 1-2: USB disconnect, device number 18
```

I thought it was a lack of power in the pi, so I plugged it simultaneously to the computer and to a power supply. The result was the same.

I tried with two raspberry pi zero V1.3 and two different usb wire with no changes.

I did those tests from two computers with same results:
```
manjaro 20.20
linux 5.8.18-1-MANJARO
```
```
linuxmint 4
LMDE 4
linux 4.19.0-12-amd64
```
I connected a monitor to the pi, it got no signal, so no boot seems to happend.
The "act" led is not blinking.

Is this issue related to udev ?
Is it related to my raspbian image or to the initrd I generated ?

burtyb, 2020-11-17:
I don't know how old the bootcode.bin is you're using but it might be worth trying the one from https://github.com/raspberrypi/usbboot/pull/64#issuecomment-707540855

adrian-amaglio, 2020-11-18:
Thanks a lot, it works !
I don’t remember copying the `bootcode.bin` file. It might be the one of the `msd` directory.

closed 2020-11-18 by adrian-amaglio