raspberrypi/usbboot #72: Selecting USB port message loop

JE-Tv1, 2021-02-10:
When I use the option to look for CMs attached to USB port, the Rpiboot continues to output "Loading embedded: bootcode.bin".
I am currently using a raspberry pi 4 B to boot a CM3+ with CMIO. 

Is there any way to not get these infinite messages and just like the usual blank waiting?

[image]

stephenglynch, 2021-05-13:
Also have this issue on windows 10 with a CM4 via USB

vanarebane, 2021-05-15:
Same issue here.

I have a 1G RAM, 8G eMMC without wifi module that works fine on the official IO board.

Then I have a 4G RAM, 8G eMMC with wifi that has the same loading loop as others here.

So only difference is the board, drivers and IO board is working

salmanfarisvp, 2021-07-16:
@vanarebane can you share how you solved the issue?

vanarebane, 2021-07-16:
@salmanfarisvp that's right, just used another port. Maybe windows assigns some drivers to specific port? I have seen some other device drivers also fixed by switching ports

salmanfarisvp, 2021-07-16:
@vanarebane Thanks 👍. I was using balena Fin v1.

CrowntailTW0608, 2021-09-06:
I've the same issue
and I removed all USB devices (including mouse) to solved the problem

lenkapolaskova, 2021-09-29:
Hello,
I have the same problem. Selecting another port does not solve the problem.

cvabc, 2021-11-26:
Sample problem here on a windows 11 pc for cm4.

timg236, 2022-03-31:
No response from OP. Closing.

N.B. We can't provide support for the multitude of Windows versions and USB versions. I recommend using a Raspberry Pi or Linux PC as the host and avoiding hubs.

closed 2022-03-31 by timg236

nerososft, 2022-04-23:
> Sample problem here on a windows 11 pc for cm4.

me too

carlonluca, 2022-07-26:
For those with the same problem in a VMWare machine, I solved by setting the USB to version 2 instead of 3 in the machine settings.

bigttys0, 2023-01-27:
Data packets sent to and from the USB port must be exactly 8 bytes in size. Any packets sent with a size that is not 8 bytes will not be read correctly by the USB port and may cause errors or data loss. To ensure reliable communication between the USB port and the device, it is important to ensure the exact size of data packet is sent.

Or, use USB2.

wangchaoHZ, 2023-04-09:
i have the same issue too，CM4 on Windows 10