raspberrypi/usbboot #73: rpiboot program encounter Failed control transfer.

yoyojacky, 2021-02-23:
I have flashed eMMC via Raspberry Pi 4B with ./rpiboot, and it shows some error , does it matter? 
`
pi@rpi8g:~/usbboot $ make
cc -Wall -Wextra -g -o bin2c bin2c.c
./bin2c msd/bootcode.bin msd/bootcode.h
./bin2c msd/start.elf msd/start.h
./bin2c msd/bootcode4.bin msd/bootcode4.h
./bin2c msd/start4.elf msd/start4.h
cc -Wall -Wextra -g -o rpiboot main.c -lusb-1.0
pi@rpi8g:~/usbboot $ sudo ./rpiboot
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Failed control transfer (-1,24)
Failed to write correct length, returned -1
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Second stage boot server
Loading embedded: start4.elf
File read: start4.elf
Second stage boot server done
`

NumberOneGit, 2021-03-02:
I'm receiving the same error with my new CM4 modules in Ubuntu and in Windows (but in Windows it replaces -1 with -9).

aethernet, 2022-01-28:
same here on mac with a CM4

timg236, 2022-03-31:
Duplicate of https://github.com/raspberrypi/usbboot/issues/36. Please can you re-test with the latest release of rpiboot.

closed 2022-03-31 by timg236