raspberrypi/usbboot #81: [Make] Libusb error

Parnoud, 2021-05-30:
Hello,

Can't make because of libusb :
```
pierrearnoud@macbook-pro-de-pierre usbboot % make
cc -Wall -Wextra -g -o rpiboot main.c -lusb-1.0
main.c:1:10: fatal error: 'libusb-1.0/libusb.h' file not found
#include <libusb-1.0/libusb.h>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

I have libusb insatlled
```
pierrearnoud@macbook-pro-de-pierre usbboot % brew info libusb
libusb: stable 1.0.24 (bottled), HEAD
Library for USB device access
https://libusb.info/
/opt/homebrew/Cellar/libusb/1.0.24 (22 files, 560.5KB) *
  Poured from bottle on 2021-05-30 at 18:06:35
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/libusb.rb
License: LGPL-2.1-or-later
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 64,482 (30 days), 211,215 (90 days), 813,259 (365 days)
install-on-request: 6,988 (30 days), 19,139 (90 days), 77,925 (365 days)
build-error: 0 (30 days)
```
I'm working on MacBook Pro : 
[image]

cuvtixo, 2021-05-31:
I'm wondering if you need libusb comment on PATH (or maybe the entire Homebrew is in path? Sorry I don't have it anymore) Libusb is in MacPorts (same 1.0.24)  so if you can install that, or possibly just build and install Libusb(assuming you know how to do this) My best guess, from memory, is that some things just don't work nice with Homebrew, and things like Libusb need to be installed "natively". Homebrew is a bit of a hack, I think it gets installed in userspace, but that means System sometimes doesn't have access. It's a non-official addition to MacOS, and being M1, this may also complicate matters for both Homebrew as well as usbboot itself. 
I'm not sure if Homebrew is the only way you know how to install Open Source tools. That seems like an obvious problem, otherwise I don't know why you would post the detection of Homebrew libusb, or why you would post here instead of on a Homebrew forum. That is a very specific use case, and I don't know if usbboot has access to such machines.

Parnoud, 2021-05-31:
Thank for your answer. Finally I just figured out how solutions my problem !
You was right, we need to include homebrew libusb : 
we can juste modify the Makefile with this : 
1. Find the libusb include : 
```
% brew ls libusb
/opt/homebrew/Cellar/libusb/1.0.24/include/libusb-1.0/libusb.h
/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib
/opt/homebrew/Cellar/libusb/1.0.24/lib/pkgconfig/libusb-1.0.pc
/opt/homebrew/Cellar/libusb/1.0.24/lib/ (2 other files)
/opt/homebrew/Cellar/libusb/1.0.24/share/libusb/ (9 files)
```
2. Add include to Makefile 
Warning do not include the "/libusb-1.0/libusb.h" and "/libusb-1.0.0.dylib"
```
rpiboot: main.c msd/bootcode.h msd/start.h msd/bootcode4.h msd/start4.h
        $(CC) -Wall -Wextra -g -o $@ $< -lusb-1.0 -I/opt/homebrew/Cellar/libusb/1.0.24/include -L/opt/homebrew/Cellar/libusb/1.0.24/lib
```
3. Make
```
% make
cc -Wall -Wextra -g -o rpiboot main.c -lusb-1.0 -I/opt/homebrew/Cellar/libusb/1.0.24/include -L/opt/homebrew/Cellar/libusb/1.0.24/lib
```
Or the main.c and MakeFile for greater durability : 
Makefile : 
```
rpiboot: main.c msd/bootcode.h msd/start.h msd/bootcode4.h msd/start4.h
        $(CC) -Wall -Wextra -g -o $@ $<  `pkg-config --cflags --libs libusb-1.0`
```
main.c : 
```
#include <libusb.h>
```

Parnoud, 2021-05-31:
same as https://github.com/raspberrypi/usbboot/pull/25

Radek-Suski, 2021-11-11:
For some reason the solution provided by @Parnoud didn't worked for me (I screwed something most likely) but I came up with an other solution. I simply launched the terminal app in Rosetta compatibility mode and it worked fine. Just in case it could help someone

thesuperzapper, 2022-01-06:
@Parnoud @Radek-Suski I have made PR https://github.com/raspberrypi/usbboot/pull/107, which fixes this issue.

closed 2022-01-06 by pelwell