raspberrypi/usbboot #87: Stuck in loop - Failed control transfer (-7,24)

ddibiasi, 2021-08-04:
Hello,

I'm trying to flash the CM4 on the official IO board.

The jumper has been set to 'Fit jumper to disable eMMC Boot' and I am using the latest usbboot.

If i try to run `sudo ./rpiboot` it gets stuck in this loop:

```
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Failed control transfer (-7,24)
Failed to write correct length, returned -7
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Failed control transfer (-7,24)
Failed to write correct length, returned -7
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Failed control transfer (-7,24)
Failed to write correct length, returned -7
```

This issue is probably related to #82.

Output of  `sudo ./rpiboot -vv`

```
Waiting for BCM2835/6/7/2711...
Found device 1 idVendor=0x1d6b idProduct=0x0003
Bus: 8, Device: 1 Path: 8
Found device 2 idVendor=0x045e idProduct=0x0719
Bus: 7, Device: 5 Path: 7-4
Found device 3 idVendor=0x03f0 idProduct=0x094a
Bus: 7, Device: 4 Path: 7-3
Found device 4 idVendor=0x17f6 idProduct=0x0862
Bus: 7, Device: 3 Path: 7-2
Found device 5 idVendor=0x05ac idProduct=0x1301
Bus: 7, Device: 2 Path: 7-1
Found device 6 idVendor=0x1d6b idProduct=0x0002
Bus: 7, Device: 1 Path: 7
Found device 7 idVendor=0x1d6b idProduct=0x0003
Bus: 6, Device: 1 Path: 6
Found device 8 idVendor=0x1d6b idProduct=0x0002
Bus: 5, Device: 1 Path: 5
Found device 9 idVendor=0x1d6b idProduct=0x0003
Bus: 4, Device: 1 Path: 4
Found device 10 idVendor=0x1d6b idProduct=0x0002
Bus: 3, Device: 1 Path: 3
Found device 11 idVendor=0x1d6b idProduct=0x0003
Bus: 2, Device: 1 Path: 2
Found device 12 idVendor=0x0a5c idProduct=0x2711
Bus: 1, Device: 8 Path: 1-6
Found candidate Compute Module...Loading embedded: bootcode4.bin
Device located successfully
Initialised device correctly
Found serial number 3
Sending bootcode.bin
libusb_bulk_transfer sent 24 bytes; returned 0
Writing 121944 bytes
libusb_bulk_transfer sent 121944 bytes; returned 0
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
Found device 1 idVendor=0x1d6b idProduct=0x0003
Bus: 8, Device: 1 Path: 8
Found device 2 idVendor=0x045e idProduct=0x0719
Bus: 7, Device: 5 Path: 7-4
Found device 3 idVendor=0x03f0 idProduct=0x094a
Bus: 7, Device: 4 Path: 7-3
Found device 4 idVendor=0x17f6 idProduct=0x0862
Bus: 7, Device: 3 Path: 7-2
Found device 5 idVendor=0x05ac idProduct=0x1301
Bus: 7, Device: 2 Path: 7-1
Found device 6 idVendor=0x1d6b idProduct=0x0002
Bus: 7, Device: 1 Path: 7
Found device 7 idVendor=0x1d6b idProduct=0x0003
Bus: 6, Device: 1 Path: 6
Found device 8 idVendor=0x1d6b idProduct=0x0002
Bus: 5, Device: 1 Path: 5
Found device 9 idVendor=0x1d6b idProduct=0x0003
Bus: 4, Device: 1 Path: 4
Found device 10 idVendor=0x1d6b idProduct=0x0002
Bus: 3, Device: 1 Path: 3
Found device 11 idVendor=0x1d6b idProduct=0x0003
Bus: 2, Device: 1 Path: 2
Found device 12 idVendor=0x0a5c idProduct=0x2711
Bus: 1, Device: 8 Path: 1-6
Found candidate Compute Module...Loading embedded: bootcode4.bin
Device located successfully
Initialised device correctly
Found serial number 3
Sending bootcode.bin
Failed control transfer (-7,24)
Failed to write correct length, returned -7
Waiting for BCM2835/6/7/2711...
Found device 1 idVendor=0x1d6b idProduct=0x0003
Bus: 8, Device: 1 Path: 8
Found device 2 idVendor=0x045e idProduct=0x0719
Bus: 7, Device: 5 Path: 7-4
Found device 3 idVendor=0x03f0 idProduct=0x094a
Bus: 7, Device: 4 Path: 7-3
Found device 4 idVendor=0x17f6 idProduct=0x0862
Bus: 7, Device: 3 Path: 7-2
Found device 5 idVendor=0x05ac idProduct=0x1301
Bus: 7, Device: 2 Path: 7-1
Found device 6 idVendor=0x1d6b idProduct=0x0002
Bus: 7, Device: 1 Path: 7
Found device 7 idVendor=0x1d6b idProduct=0x0003
Bus: 6, Device: 1 Path: 6
Found device 8 idVendor=0x1d6b idProduct=0x0002
Bus: 5, Device: 1 Path: 5
Found device 9 idVendor=0x1d6b idProduct=0x0003
Bus: 4, Device: 1 Path: 4
Found device 10 idVendor=0x1d6b idProduct=0x0002
Bus: 3, Device: 1 Path: 3
Found device 11 idVendor=0x1d6b idProduct=0x0003
Bus: 2, Device: 1 Path: 2
Found device 12 idVendor=0x0a5c idProduct=0x2711
Bus: 1, Device: 8 Path: 1-6
Found candidate Compute Module...Loading embedded: bootcode4.bin
Device located successfully
Initialised device correctly
Found serial number 3
Sending bootcode.bin
Failed control transfer (-7,24)
Failed to write correct length, returned -7
```

ddibiasi, 2021-08-05:
I just tried flashing again with another CM4 & the same IO board and it works as it should. 

The faulty CM4 has already an OS installed which boots if the jumper is removed, so I think this is some sort of software issue. 

Do you have any solution for this problem?

bhuvanchandra, 2021-09-15:
I'm trying the rpiboot for Rpi CM3 module. I see similar errors:

with `sudo ./rpiboot `
$ sudo ./rpiboot
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode.bin
Sending bootcode.bin
Failed to write correct length, returned 0

and with `sudo ./rpiboot -d msd`
sudo ./rpiboot
Loading: msd/bootcode.bin
Loading: msd/bootcode4.bin
Waiting for BCM2835/6/7/2711...
Loading: msd/bootcode.bin
Sending bootcode.bin
Failed to read correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: msd/bootcode.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: msd/bootcode.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: msd/bootcode.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: msd/bootcode.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: msd/bootcode.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: msd/bootcode.bin
Sending bootcode.bin
Failed to write correct length, returned 0

ddibiasi, 2021-09-15:
Is there any progress on this issue, as this seems to affect multiple people and potentially bricks CM4s?

scott-williams-xentronics, 2021-11-11:
I had this issue, but it was on a fresh Compute Module 4.
Tried it on another PC (which was also running an older version of rpiboot) and it worked fine.
So, in my case, it was either a (1) PC host specific issue, or (2) something wrong with a recent update of this Git repo.

ddibiasi, 2021-11-11:
This issue still exists despite having tried multiple different USB cables, as well as power supplies, as mentioned by @Sagar73594  .

I also have tried usbboot on multiple different PCs, using different operating systems (Linux, Windows) to no avail, as suggested by @scott-williams-xentronics  . 

As this was last tried in August, there could potentially have been a fix in the GitHub repo, so I'll need to retry usbboot with the said bricked device.

dir-ableton, 2021-11-23:
I'm also running into this. I tried several cables, ports, hubs and no hubs in between, several CM4s and two different IO boards. The problem persists. Using a USB protocol analyzer I see nothing of significance (but I am not an expert on this at all).

However I just plugged the exact same IO board + CM4 + cable to a different computer - and there it works.

The notebook where it's not working is an DELL XPS and the XHCI controller is named 

  `Tiger Lake-H USB 3.2 Gen 2x1 xHCI Host Controller `

according to https://pci-ids.ucw.cz/read/PC/8086/43ed

The PC where it works lists the respective USB host controller as 

  `01:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] X399 Series Chipset USB 3.1 xHCI Controller (rev 02)`

Probably not something anybody here could do anything about, but OTOH maybe a puzzle piece to understand the issue better.

scott-williams-xentronics, 2021-11-24:
Super interesting. 
I am having the issue on a Dell XPS 17, brand new.
Must be some common issue with the USB Controller.

I wonder if there's any action we can take?

dir-ableton, 2021-11-24:
I have some new information in the meantime. While it appeared to work (meaning the rpiboot was actually doing what it's supposed to do), mounting and working with the block device is fraught with errors. It produces fun error messages, see below.

Using a newer Dell XPS of a colleague with USB host controllers identifying as "something something Thunderbolt 4" (sorry, I just read the output of lsusb and recite from memory) actually did work for rpiboot, but again mounting would fail.

This whole thing is driving me nuts.

```
[  716.896000] usb 1-10: new high-speed USB device number 3 using xhci_hcd
[  717.059073] usb 1-10: config index 0 descriptor too short (expected 55, got 32)
[  717.074976] usb 1-10: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[  717.074982] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  717.074985] usb 1-10: Product: BCM2711 Boot
[  717.074987] usb 1-10: Manufacturer: Broadcom
[  717.074989] usb 1-10: SerialNumber: 9c45210c
[  719.458137] usb 1-10: USB disconnect, device number 3
[  719.732014] usb 1-10: new high-speed USB device number 4 using xhci_hcd
[  719.911348] usb 1-10: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[  719.911354] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[  719.911357] usb 1-10: Product: BCM2711 Boot
[  719.911359] usb 1-10: Manufacturer: Broadcom
[  719.911361] usb 1-10: SerialNumber: 9c45210c
[  721.511036] usb 1-10: USB disconnect, device number 4
[  723.204084] usb 1-10: new high-speed USB device number 5 using xhci_hcd
[  723.379801] usb 1-10: New USB device found, idVendor=0a5c, idProduct=0001, bcdDevice= 0.01
[  723.379807] usb 1-10: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[  723.379810] usb 1-10: Product: Compute Module
[  723.379812] usb 1-10: Manufacturer: Raspberry Pi
[  723.379814] usb 1-10: SerialNumber: 9c45210c
[  723.407274] usb-storage 1-10:1.0: USB Mass Storage device detected
[  723.407523] scsi host10: usb-storage 1-10:1.0
[  723.407660] usbcore: registered new interface driver usb-storage
[  723.409912] usbcore: registered new interface driver uas
[  724.416325] scsi 10:0:0:0: Direct-Access     RPi-MSD- 0001                  PQ: 0 ANSI: 2
[  724.417080] scsi 10:0:0:0: Attached scsi generic sg0 type 0
[  724.417515] sd 10:0:0:0: [sda] 62333952 512-byte logical blocks: (31.9 GB/29.7 GiB)
[  724.417656] sd 10:0:0:0: [sda] Write Protect is off
[  724.417659] sd 10:0:0:0: [sda] Mode Sense: 0f 00 00 00
[  724.417803] sd 10:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[  724.450049]  sda: sda1 sda2 sda3 sda4
[  724.470161] sd 10:0:0:0: [sda] Attached SCSI removable disk
[  737.959173] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[  740.596673] sd 10:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_SENSE cmd_age=0s
[  740.596690] sd 10:0:0:0: [sda] tag#0 Sense Key : Hardware Error [current] 
[  740.596694] sd 10:0:0:0: [sda] tag#0 Add. Sense: No additional sense information
[  740.596704] sd 10:0:0:0: [sda] tag#0 CDB: Write(10) 2a 00 00 01 42 08 00 00 18 00
[  740.596712] blk_update_request: I/O error, dev sda, sector 82440 op 0x1:(WRITE) flags 0x800 phys_seg 3 prio class 0
[  740.596721] Buffer I/O error on dev sda2, logical block 65, lost async page write
[  740.596728] Buffer I/O error on dev sda2, logical block 66, lost async page write
[  740.596732] Buffer I/O error on dev sda2, logical block 67, lost async page write
[  740.680490] sd 10:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_SENSE cmd_age=0s
[  740.680502] sd 10:0:0:0: [sda] tag#0 Sense Key : Hardware Error [current] 
[  740.680506] sd 10:0:0:0: [sda] tag#0 Add. Sense: No additional sense information
[  740.680510] sd 10:0:0:0: [sda] tag#0 CDB: Write(10) 2a 00 00 01 42 28 00 00 10 00
[  740.680513] blk_update_request: I/O error, dev sda, sector 82472 op 0x1:(WRITE) flags 0x800 phys_seg 2 prio class 0
[  740.680520] Buffer I/O error on dev sda2, logical block 69, lost async page write
[  740.680526] Buffer I/O error on dev sda2, logical block 70, lost async page write
[  740.760463] sd 10:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_SENSE cmd_age=0s
[  740.760477] sd 10:0:0:0: [sda] tag#0 Sense Key : Hardware Error [current] 
[  740.760481] sd 10:0:0:0: [sda] tag#0 Add. Sense: No additional sense information
[  740.760485] sd 10:0:0:0: [sda] tag#0 CDB: Write(10) 2a 00 00 01 42 78 00 00 08 00
[  740.760488] blk_update_request: I/O error, dev sda, sector 82552 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
[  740.760495] Buffer I/O error on dev sda2, logical block 79, lost async page write
[  740.840534] sd 10:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_SENSE cmd_age=0s
[  740.840539] sd 10:0:0:0: [sda] tag#0 Sense Key : Hardware Error [current] 
[  740.840542] sd 10:0:0:0: [sda] tag#0 Add. Sense: No additional sense information
[  740.840545] sd 10:0:0:0: [sda] tag#0 CDB: Write(10) 2a 00 00 01 42 88 00 00 10 00
[  740.840546] blk_update_request: I/O error, dev sda, sector 82568 op 0x1:(WRITE) flags 0x800 phys_seg 2 prio class 0
[  740.840552] Buffer I/O error on dev sda2, logical block 81, lost async page write
[  740.840557] Buffer I/O error on dev sda2, logical block 82, lost async page write
[... cut ...]
[  967.612345]  ? process_one_work+0x3c0/0x3c0
[  967.612348]  kthread+0x12b/0x150
[  967.612351]  ? set_kthread_struct+0x40/0x40
[  967.612354]  ret_from_fork+0x22/0x30
[  967.612539] INFO: task scsi_eh_10:9127 blocked for more than 120 seconds.
[  967.612542]       Not tainted 5.11.0-40-generic #44~20.04.2-Ubuntu
[  967.612544] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  967.612545] task:scsi_eh_10      state:D stack:    0 pid: 9127 ppid:     2 flags:0x00004000
[  967.612549] Call Trace:
```

timg236, 2021-11-24:
The recommended setup for the usbboot host is

- Raspberry Pi OS on a Raspberry Pi 4 (or 3)
- Git clone the latest software and build from source
- Use a direct USB cable connection and not a hub

libusb on Windows seems to be a bit flakey and doesn't always discover devices behind hubs. Since there's an infinite variety of device OS/USB configurations we only test with the above.

dir-ableton, 2021-11-24:
Thank you for the clarification. And a good idea for a workaround if I can't get my Linux XHCI controllers to behave :)

dir-ableton, 2021-12-02:
Ok I tried this - PI3, CM4 IO Board, latest Pi Os, cloned & buillt rpiboot on the system, and got essentially the same result I see everywhere else. See dmesg output below.

My interpretation here is that the writing is stuck somehow and the kernel worker queues responsible start complaining if the tasks they were given aren't finished.

I tested this for now just with one setup. However I did test several CM4 modules, SD-cards and cables with the same results, albeit with a PC host. I can conduct the same tests with the PI3, but am at this point a bit skeptical about it yielding different resutlts.

```
  83.667980] EXT4-fs (mmcblk0p2): resized filesystem to 7566848
[  996.281656] usb 1-1.4: USB disconnect, device number 4
[  996.284307] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[  996.284654] sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x01 driverbyte=0x00
[  999.135258] usb 1-1.4: new high-speed USB device number 5 using dwc_otg
[  999.265681] usb 1-1.4: config index 0 descriptor too short (expected 55, got 32)
[  999.266202] usb 1-1.4: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[  999.266218] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  999.266231] usb 1-1.4: Product: BCM2711 Boot
[  999.266243] usb 1-1.4: Manufacturer: Broadcom
[  999.266256] usb 1-1.4: SerialNumber: f06348ef
[  999.268494] pcm512x 1-004d: supply AVDD not found, using dummy regulator
[  999.268700] pcm512x 1-004d: supply DVDD not found, using dummy regulator
[  999.268776] pcm512x 1-004d: supply CPVDD not found, using dummy regulator
[ 1005.461342] pcm512x 1-004d: supply AVDD not found, using dummy regulator
[ 1005.461616] pcm512x 1-004d: supply DVDD not found, using dummy regulator
[ 1005.461713] pcm512x 1-004d: supply CPVDD not found, using dummy regulator
[ 1006.267678] usb 1-1.4: USB disconnect, device number 5
[ 1009.375251] usb 1-1.4: new high-speed USB device number 6 using dwc_otg
[ 1009.505701] usb 1-1.4: config index 0 descriptor too short (expected 55, got 32)
[ 1009.506208] usb 1-1.4: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[ 1009.506225] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1009.506238] usb 1-1.4: Product: BCM2711 Boot
[ 1009.506250] usb 1-1.4: Manufacturer: Broadcom
[ 1009.506263] usb 1-1.4: SerialNumber: f06348ef
[ 1009.508429] pcm512x 1-004d: supply AVDD not found, using dummy regulator
[ 1009.508611] pcm512x 1-004d: supply DVDD not found, using dummy regulator
[ 1009.508698] pcm512x 1-004d: supply CPVDD not found, using dummy regulator
[ 1010.553677] pcm512x 1-004d: supply AVDD not found, using dummy regulator
[ 1010.553931] pcm512x 1-004d: supply DVDD not found, using dummy regulator
[ 1010.554026] pcm512x 1-004d: supply CPVDD not found, using dummy regulator
[ 1012.005433] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[ 1012.005467] usb 1-1: USB disconnect, device number 2
[ 1012.005488] usb 1-1.1: USB disconnect, device number 3

[ 1012.006400] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 3
[ 1012.006433] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet
[ 1012.006817] smsc95xx 1-1.1:1.0 eth0: Link is Down
[ 1012.006847] smsc95xx 1-1.1:1.0 eth0: Failed to read reg index 0x00000114: -19
[ 1012.006863] smsc95xx 1-1.1:1.0 eth0: Error reading MII_ACCESS
[ 1012.006878] smsc95xx 1-1.1:1.0 eth0: __smsc95xx_mdio_read: MII is busy
[ 1012.006923] smsc95xx 1-1.1:1.0 eth0: Failed to read reg index 0x00000114: -19
[ 1012.006938] smsc95xx 1-1.1:1.0 eth0: Error reading MII_ACCESS
[ 1012.006953] smsc95xx 1-1.1:1.0 eth0: __smsc95xx_mdio_read: MII is busy
[ 1012.006994] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[ 1012.071161] usb 1-1.4: USB disconnect, device number 6
[ 1012.275373] Indeed it is in host mode hprt0 = 00001501
[ 1012.485235] usb 1-1: new high-speed USB device number 7 using dwc_otg
[ 1012.485327] Indeed it is in host mode hprt0 = 00001101
[ 1012.725418] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
[ 1012.725432] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1012.725930] hub 1-1:1.0: USB hub found
[ 1012.725996] hub 1-1:1.0: 5 ports detected
[ 1012.727648] pcm512x 1-004d: supply AVDD not found, using dummy regulator
[ 1012.727775] pcm512x 1-004d: supply DVDD not found, using dummy regulator
[ 1012.727829] pcm512x 1-004d: supply CPVDD not found, using dummy regulator
[ 1013.045250] usb 1-1.1: new high-speed USB device number 8 using dwc_otg
[ 1013.185545] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
[ 1013.185563] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1013.188433] smsc95xx v2.0.0
[ 1013.261600] libphy: smsc95xx-mdiobus: probed
[ 1013.262806] pcm512x 1-004d: supply AVDD not found, using dummy regulator
[ 1013.262993] pcm512x 1-004d: supply DVDD not found, using dummy regulator
[... 140 lines cut ...]
[ 1472.477250]  r10:00000000 r9:80f05008 r8:00000bb8 r7:85883800 r6:00000005 r5:00000000
[ 1472.477263]  r4:85857e7c
[ 1472.477289] [<8074c750>] (scsi_test_unit_ready) from [<80761844>] (sd_check_events+0x114/0x158)
[ 1472.477310]  r9:834c032c r8:000000c0 r7:bab0af00 r6:85883800 r5:80f05008 r4:829b4000
[ 1472.477332] [<80761730>] (sd_check_events) from [<806022fc>] (disk_check_events+0x5c/0x180)
[... cut ...]
[ 1472.477565] [<80143620>] (kthread) from [<801000ec>] (ret_from_fork+0x14/0x28)
[ 1472.477579] Exception stack(0x85857fb0 to 0x85857ff8)
[ 1472.477596] 7fa0:                                     00000000 00000000 00000000 00000000
[ 1472.477615] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1472.477633] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 1472.477654]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80143620
[ 1472.477667]  r4:84c7ed80
pi@raspberrypi:~ $ 
```

timg236, 2021-12-02:
I think the callstack indicates that the a write command never completed so the kernel timed out and killed the process. How are you writing the image to the Compute Module e.g .imager or are you just accessing files directly?

dir-ableton, 2021-12-06:
I've been using both dd to flash the whole image in this case, but also observed problems in the past when mounting and accessing, especially again writing.

dir-ableton, 2021-12-07:
A quick follow-up: it appears the SD-card-brand makes a difference. I got eMMC CMs today, and with them I was able to flash. I then used a SanDISK SD-card in the IO board + CM4 that was failing before, and have flashed 20 or so times in a row, no problem. Back to the NetAC-card I used, and I get sporadic failures.

The speed seems to have degraded between CM3 and CM4, it used to be between 5-6MB/s, now it's about 3.5-4, 4.7 with eMMC. 

Is there a chance of increasing compatibility or speed?

timg236, 2022-03-31:
@ddibiasi Please could you retry with the latest version. Over the last few months there's been some improvements to how the USB control is inherited from the ROM so it would be useful to know if that helps.

N.B. The recommended host is Raspberry Pi or Linux PC without a hub between the Compute Module and the host.

brooksphilip, 2022-05-15:
I have this same problem on an M1 MacBook Pro. I can not get rpiboot to work. 

Found candidate Compute Module...Loading embedded: bootcode4.bin
Device located successfully
Initialised device correctly
Found serial number 3
Sending bootcode.bin
libusb_bulk_transfer sent 24 bytes; returned 0
Writing 121888 bytes
libusb_bulk_transfer sent 121888 bytes; returned 0
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
Found device 1 idVendor=0x0a5c idProduct=0x2711
Bus: 1, Device: 1 Path: 1-1
Found candidate Compute Module...Loading embedded: bootcode4.bin
Device located successfully
Initialised device correctly
Found serial number 3
Sending bootcode.bin
libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
Failed control transfer (-1,24)
Failed to write correct length, returned -1
libusb: warning [darwin_close] USBDeviceClose: no connection to an IOService
Waiting for BCM2835/6/7/2711...

boedy, 2022-08-24:
I'm running into the same issue. I received a new batch of 10 CM4's that are all experiencing this issue. I've compared these units with working ones and noticed some visual differences between them. I've included a picture of the two below. I've marked the differences on the unit that's having this issue. This seems to be the newer device.

[image]

jonrot1906, 2022-08-26:
> I'm running into the same issue. I received a new batch of 10 CM4's that are all experiencing this issue. I've compared these units with working ones and noticed some visual differences between them. I've included a picture of the two below. I've marked some differences on the unit that's having this issue. This seems to be the newer device.
> 
> [image]
> 
> **Edit** I discovered the underlying issue. In my instance seems to be related to [Balena Etcher not yet supporting the newer CM4's](https://github.com/balena-io/etcher/issues/3792). When I tried running `rpiboot` Balena Etcher was still open in the background, which wouldn't allow `rpiboot` to connect as it seems to keep the device busy. Closing Etcher and re-running `rpiboot` allowed it to mount to my macbook.

Great finding. This was also the case for me - I was just a second to giving up on this CM4 and then saw your edit. Balena Etcher seemed to block a correct connection as well. I had to reattach the USB cable to my mac, rerun `rpiboot` and everything worked fine.

fix3000, 2022-09-01:
Yep, I also found that I had to update rpiboot to a newer version in order to handle the newer CM4 hardware I received. And unfortunately, you cannot run rpiboot and balenaEtcher simultaneously. You have to do rpiboot first, then bE.

Hopefully they'll fix this detection issue soon... I get the sense that a lot of folks were forced to the CM4 due to the CM3+ supply issues. Manufacturing is a complete cluster at the moment.

JamesH65, 2022-09-02:
New devices have been sent to balena for verification of use with etcher, however, I've not had confirmation from them that it has yet been done, although it was a while ago. I'll email them again to see if there have been any progress.

timg236, 2022-09-02:
The new CM4s report identical USB descriptors and the SoC is the same, there are no differences in the the USB interface. All that's required is to have a new enough version of rpiboot so that it loads firmware supporting the new PMIC.

The significant difference is not circled in the photos. See bottom left, the MXL power supply has been replaced with a Dialog part.

capiman2, 2022-09-02:
@timg236 
Can you tell what happens when old rpiboot and new PMIC is used together? Is there a check and it fails?
What version supports both?
Is there some kind of comparison between old and new CM4? Any change in behaviour in general?
Is there a possibility to check if the one or the other is used? Some program which can be called to get this info?
Is there any possibility to buy/get one CM4 (Wifi, no MMC) with Dialog Part, so we can test/confirm our product works also with the dialog part?

timg236, 2022-09-02:
The latest version of rpiboot supports both but I think it's been supported for well over a year now.
If the firmware used by rpiboot is too old then it will fail to boot - maybe an error pattern but the rpiboot host won't know anything about it.
I can't advise about ordering CM4 parts for test.

The Product Information Portal is the place to go for product update information
https://pip.raspberrypi.com/

JamesH65, 2022-09-02:
Please email business@raspberrypi.com to request parts for testing, please give as much information as possible on your use case, so our commercial team can make the appropriate decisions on supply. However, there is no guarantee parts are available at this time.

capiman2, 2022-09-02:
https://pip.raspberrypi.com/categories/645/documents/RP-001606-PC/RP-001606-PC-2.pdf

"
In software, new product can be distinguished by:
cat /proc/cpuinfo will report "Model" as "Raspberry Pi Compute Module 4 Rev 1.1"
"

JamesH65, 2022-09-02:
> https://pip.raspberrypi.com/categories/645/documents/RP-001606-PC/RP-001606-PC-2.pdf
> 
> " In software, new product can be distinguished by: cat /proc/cpuinfo will report "Model" as "Raspberry Pi Compute Module 4 Rev 1.1" "

Are you reporting this as an error in the PCN? Or just reporting it?

capiman2, 2022-09-02:
"Are you reporting this as an error in the PCN? Or just reporting it?"

Just for info, that above info can be found in an official document

NeusAap, 2023-04-05:
I had this problem as well with a CM4 on the official IO-Board. 
I was using a (analog/old-school) variable power supply, which was set to 12.5V by accident.

Dialing it down to 12V exactly, fixed the problem for me. Might be worth a try for other people using variable power supplies.

JaggerJo, 2023-09-04:
Had the same issue on a M2 Mac mini. Swapping USB cables worked.

Jatin-Patel2331, 2023-09-19:
> Hello,
>
> I'm trying to flash the CM4 on the official IO board.
>
> The jumper has been set to 'Fit jumper to disable eMMC Boot' and I am using the latest usbboot.
>
> If i try to run `sudo ./rpiboot` it [...]

I had the same error, reinstalling the "rpiboot" worked for me.

ll-O-ll, 2024-06-03:
I get the same loop when running the `sudo ./rpiboot -d recovery` in the [secure-boot-example](https://github.com/raspberrypi/usbboot/tree/master/secure-boot-example)

ddibiasi, 2025-02-06:
Chiming in with a slight delay, I haven't faced this exact issue since 2023. I assume some fixes (i.e. https://github.com/raspberrypi/usbboot/issues/87#issuecomment-1084219943) have resolved this issue for me.

As most of the problems boil down to using a proper power supply, as well as good cables, it would probably make sense to add these points to the "Troubleshooting" section for the CM4 in the main readme.

nychyporuk, 2025-02-12:
The issue still exists

jcaptitude, 2025-04-07:
I have the same problem for my CM5, tried with Windows host or Pi 5 host, neither worked.

Also tried with multiple different cable and even a dongle to  supply additional power, it still doesn't work.

But when I tried to flash my Pi 5, it worked

yoyojacky, 2026-09-01:
> > I try to use my Raspberry Pi 5B 8G and install usbboot. using Raspberry Pi Compute module 4 IO board with MicroUSB cable and met the same problem, but I have solved it by adding : usb_max_current_enable=1 to /boot/firmware/config.txt file and reboot my Raspberry pi 5. it works fine~
> 
> this work for CM4 ??

Yes, it's the same~

timg236, 2026-10-28:
This issue is a confusing mix of different problems, mostly power supply / USB cables etc. Closing.

closed 2026-10-28 by timg236

julianoes, 2026-10-28:
@timg236 is what I described in https://github.com/raspberrypi/usbboot/issues/87#issuecomment-4608321882 fixed?