raspberrypi/usbboot #90: [question] Possible to make rpiboot return the /dev/sd? or find it otherwise - with certinty

hippyau, 2021-08-13:
Hi there, 

 I want to script an image update process for my CM4, and hoping you can point out how to get the `/dev/sd?` that will appear after `./rpiboot` is executed. 

Can it be returned somehow, or have a shell method to find it reliably?   I'd hate for another drive to be added ad-hoc to the machine and accidentally it's overwritten by a hardcoded `/dev/sdd` in my `dd` command.

Cheers

timg236, 2021-08-13:
Not really, there's no handshake between rpiboot and the Pi because rpiboot is just loading some arbitrary firmware to the device. When the msd firmware runs it disconnects from rpiboot and appears as a different device.

You might be able to tweak your scripts to look under
/sys/class/block/device/sd**a**/device and check if the vendor contains 'RPi-MSD'

woju, 2022-03-24:
There's `/dev/disk/by-id/usb-RPi-MSD-_0001_????????-0:0`.

timg236, 2022-03-31:
Yes, you can write u-dev rules to do this. However, I think this out of scope for rpiboot.

The OP's original concern about dd can be addressed by using the Raspberry Pi Imager to flash images which goes to some lengths to avoid flashing the local OS drive.

N.B The new mass-storage-gadget mode provides more descriptive makes in the MSD descriptors (e.g. NVMe / mmcblk0)
https://github.com/raspberrypi/usbboot/blob/master/mass-storage-gadget/README.md

closed 2022-03-31 by timg236