raspberrypi/usbboot #99: Unable to generate valid boot.img using tools/make-boot-image

Ericj92, 2021-11-11:
I am facing to problem with the make-boot-image tool.
My final goal would be to generate a boot.img which I can upload using the rpiboot.exe tool.
First of all I started with the files from the imager subdirectory, I mounted the /imager/boot.img file to checkout the required files and configurations. I then copied the content of the boot.img file, using my mount point, to a new directory called ../secure-boot-files.
I then unmounted the /imager/boot.img file.
Next I called the following command for the usbboot toplevel directory:
sudo ./tools/make-boot-image -d secure-boot-files -o boot.img

The result is a 22MB boot.img file, while the original /imager/boot.img file has a filesize of approx. 42MB. 
When I mount the 22MB boot.img file I get an empty directory, while mounting the /imager/boot.img gave me an list of familiar files(*.dtb etc). 

What went wrong in my setup, for the files not to show up in the generated boot.img file?

timg236, 2021-11-11:
It's not possible to tell from your description. Try adding 'set -x' to the start of the make-boot-image script to see what's happening

Ericj92, 2021-11-11:
I added the 'set -x' above the 'set -e' command and got a whole lot of debug when I ran the command:
sudo ./tools/make-boot-image -d boot-yocto -o boot.img

I have attached a file with the output of this command, hopefully it is to any use for you.
[output.txt](https://github.com/raspberrypi/usbboot/files/7520836/output.txt)

Ericj92, 2021-11-11:
Thanks, the make-boot-image from the PR fixed the issue. I can now reproduce the boot.img file. Next for me is to replace the zImage with a u-boot binary and later on a u-boot binary, kernel and initramfs. But for now thank you for your quick response

closed 2021-11-13 by timg236