raspberrypi/usbboot #101: Support for Pi Zero 2

Janghou, 2021-11-12:
Trying out but not working:

Error:
[code]Sending bootcode.bin
Failed control transfer (-7,24)
Failed to write correct length, returned -7
Waiting for BCM2835/6/7/2711.[/code]..

Or is the new Zero 2 incompatible?

burtyb, 2021-11-12:
For me (and others) some Pi Zero 2 W work OK but others don't.

pelwell, 2021-11-12:
There is no fundamental reason why it can never work, and if it's working for you now then that's great, but we've done nothing to make sure that it is working on Zero 2 W now. For what it's worth, I saw a problem when I tested it.

At some point we may get round to understanding and fixing the issue, but for now it isn't a priority.

Lewiscowles1986, 2021-11-13:
One thing I immediately noticed is the error message does not contain BCM2710A1 will 2711 be subtly different from 2710A1? Maybe is my instinct.

I'd guess the bootcode.bin needs to be built or sourced for this model specifically, or from a time where this model is included. That might overcome. I've been thinking about this all week, so I'm glad I came here instead of wasting a day.

JamesH65, 2021-11-14:
The 2711 is very different to the 2710. Different Arm cores, different videocore. Is the difference between the Pi3 and the Pi4.

Lewiscowles1986, 2021-11-14:
Does anyone know if the bootcode.bin can be rebuilt, or if we basically wait for raspberry pi foundation to release a compatible one?

JamesH65, 2021-11-14:
It's closed source.

pelwell, 2021-11-19:
This issue stems from a bug in the BCM2837A0 ROM, which can't be modified. The effect of the bug is to stall the USB boot sequence for random (but usually long) periods while changing GPIOs and generally being a nuisance, but sometimes it does actually work - it all depends on precisely what random noise is present in boot RAM when the power is applied.

Fortunately there is an OTP bit that can be set to avoid the problem, but that can't be set via rpiboot unless you can download code. The plan is to push an update to the regular firmware that patches the OTP of Zero 2 Ws - running this once would be sufficient to fix the device forever.

Lewiscowles1986, 2021-11-20:
@pelwell any chance of a bump reminder once the firmware gets released? 🙏  or an official resource for us to watch?

pelwell, 2021-11-20:
I'll try - feel free to remind me if you've heard nothing by the end of next week.

pelwell, 2021-11-25:
It's been longer than usual since the last official firmware release - you can download a trial start.elf + fixup.dat here: https://drive.google.com/file/d/1t4HDvqq1orrGOAUdjjPcZWlKjQuWAiTU/view?usp=sharing
To be clear, although it's only necessary to run this version of start.elf once on each board, the fix will be a standard part of all future firmwares to ensure that the fix gets applied to all Zero 2 Ws.

Lewiscowles1986, 2021-11-25:
I'll try it tonight thanks 👍

burtyb, 2021-11-26:
I've had over 400 successful reboots with rpiboot on my HAT with 4 Pi Zero 2 W so it's looks good here, thanks.

Lewiscowles1986, 2021-11-26:
I think my problems could just be windows being a doof then. I've not yet used my Linux machine or mac

pelwell, 2021-12-01:
There's a fress firmware with the MSD fix in the rpi-firmware repo, as used by rpi-update.

timg236, 2022-03-31:
I think this is now resolved. Closing.

closed 2022-03-31 by timg236