raspberrypi/usbboot #105: booting imager hangs on cm4

Sfinx, 2021-12-15:
CM4 is 8GB RAM / 32GB eMMC, BT + WiFi. Hangs at

...
MESS:00:00:11.273811:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:11.281207:0: uart: Baud rate change done...
MESS:00:00:11.283228:0:
...

Log attached

[cm4.log](https://github.com/raspberrypi/usbboot/files/7722986/cm4.log)

Seems like console uart is conflicting with BT, but disabling BT still not boot the kernel.

BTW: is there way to change resolution when it will boot - for ex. I'm having 1280x800 HDMI display from waveshare

timg236, 2021-12-16:
Works for me on a CM4 I/O board. 

Although, the imager should really be deleted from here

Sfinx, 2021-12-16:
> Works for me on a CM4 I/O board.
> 
> Although, the imager should really be deleted from here

What the CM4 version ? Does it have BT ?

timg236, 2021-12-16:
I don't think it matters but you could try editing removing uart_2ndstage=1 in the config.txt or changing UART settings in the boot.img.

This isn't really part of usbboot and it's a very early beta/snapshot so I'm removing it. It probably deserves it's own repo that's updated from the imager builds.
https://github.com/raspberrypi/usbboot/pull/106

Sfinx, 2021-12-17:
BTW: Where the boot_ramdisk option bugs have to be reported ?

closed 2021-12-17 by Sfinx