raspberrypi/usbboot #116: eMMC won't boot after program_pubkey

3dlabsio, 2022-02-19:
After successfully testing a signed boot.img/boot.sig with SIGNED_BOOT=1 in boot.conf, committing the public key to OTP permanently results in eMMC not booting anymore.

I'm using the latest version from this repo (2/08/2022)

The cm4 green light yields 4 rapid blinks indicating start*.elf cannot be found.

Placing the same boot.img/boot.sig on a USB thumbdrive successfully boots via USB boot, so it doesn't appear to be a signing issue.

I have tried multiple things such as recreated the FAT32 boot partition, running fsck on it to repair it, ensuring it's properly labeled as "boot" with fat label -- nothing works, it seems as if the system cannot see or access /boot at all. 

The recovery and USB msd signed images both also work via rpiboot, so it appears to be directly limited to the emmc not being seen.

3dlabsio, 2022-02-19:
Forgot to include the relevant output from UART:

RPi: BOOTLOADER release VERSION:2be30cb4 DATE: 2022/02/16 TIME: 21:59:51 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1645048791 0x03042faa 0x00a03140 0x00478ea1                                                               
PM_RSTS: 0x00001000                                                                                                                                                                                                             
part 00000000 reset_info 00000000
uSD voltage 3.3V

Boot mode: SD (01) order 254
NETCONSOLE 6665@172.168.1.8,6666:172.168.1.7/00:00:00:00:00:00
GENET: RESET_PHY
PHY ID 600d 84a2
NET_CONSOLE: wait for link
LINK STATUS: speed: 1000 full duplex
Link ready
GENET START: 64 16 32
GENET: UMAC_START 0xdca632e9 0xdb3f0000
NET 0.0.0.0 0.0.0.0 gw 0.0.0.0 tftp 0.0.0.0
RX: 0 IP: 0 IPV4: 0 MAC: 0 UDP: 0 UDP RECV: 0 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
OCR c0ff8080 [0]
CID: 00150100414a54443452069718bc0487
Failed to open device: 'sdcard' (cmd 83a0010 status 0)
Retry SD 1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
OCR c0ff8080 [0]
CID: 00150100414a54443452069718bc0487
Failed to open device: 'sdcard' (cmd 83a0010 status 0)

timg236, 2022-02-21:
I can reproduce this, weirdly this works fine on the lite with sdcard

timg236, 2022-02-21:
@3dlabsio Please could you give the attached EEPROM a try. It looks as though the RSA signature checks in secure-boot OTP was triggering an issue in the EMMC setup that caused it to take a long time to respond to CSD. 

[pieerom-emmc-secure-boot-fix.zip](https://github.com/raspberrypi/usbboot/files/8108890/pieerom-emmc-secure-boot-fix.zip)

3dlabsio, 2022-02-21:
@timg236 This is confirmed working. Thanks for the quick fix. On a side note, it appears that config.txt still needs to be present in /boot or USB ports didn't get initialized on my system.

timg236, 2022-02-22:
By "/boot" are you referring to the file-system contained within boot.img? If so, then yes, you need to have `config.txt` file with either `otg_mode=1` (preferred) or `dtoverlay=dwc2,dr_mode=host` because by default in the USB PHY is not enabled.
The Bullseye OS image contains the following in the default `config.txt` file.

```
[cm4]
otg_mode=1
```

pelwell, 2022-02-22:
Note that if the only config.txt setting required is `otg_mode=1` then, depending on your application, you may prefer to set that semi-permanently in the EEPROM config:
```
[config.txt]
otg_mode=1
```

closed 2022-02-23 by timg236