raspberrypi/usbboot #118: Installer for Windows does not install the drivers but finishes installing rpi boot

randomhousegir, 2022-02-22:
It says it successfully extracted the drivers but then it does not install them. I am running as administrator but when I try to use rpi boot it does not connect or allow USB access.
[image]
[image]
[image]
 I am unsure how to get the drivers to install so I can set up an image on the emmc

timg236, 2022-02-23:
You probably need to wait a long time for the installer to close and then reboot. Although, since the Windows version is not actively supported it's better run this on X86 Linux or a Raspberry Pi.

timg236, 2022-03-31:
I retested on my Windows PC after uninstalling the driver and it does complete. It takes long time for the driver registration to complete so be careful to not close the command window which appears.

closed 2022-03-31 by timg236