raspberrypi/usbboot #119: rpiboot won't see NVME drives

klslz, 2022-02-24:
I ran latest rpiboot against a CM4 on a Waveshare-IO-BASE-A module.

1. uSD card is working fine - as expected

2. The NVME fs /dev/nvme0n1p1 and /dev/nvme0n1p2 won't be recognised though.

Is there anything I can do about it. Or it is simply not supported?

This to me would be a major drawback. 

No simple backups. 
No simple corrections/edits.
eeprom update would also require an uSD card to be inserted.

timg236, 2022-02-24:
It's not supported because the firmware does not have write support for NVMe. The USB-MSD firmware is a pretty hold legacy stack so unlikely to be improved.

However, the Raspberry Pi Imager embedded version can be launched via rpiboot (you need the very latest release) and that supports NVMe. I think the future direction is to load a Linux RD which can then use standard Linux commands to modify the file-system e.g. HTTP download etc or use a USB gadget driver from Linux.

klslz, 2022-02-24:
I am wondering if the rpi-imager is the right tool for day2day maintenance tasks.

The way I'd see it. You'd basically would have to disassemble the whole device or you'd  need to insert and boot an uSD to run this or that maintenance task.

However. That's the way it is. 

Would be nice to see a note about that IMO serious limitation in the otherwise excellent RPI online doc:

https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/boot-nvme.adoc

Otherwise more people will run into this trap.

THX

timg236, 2022-02-24:
It's more likely to be a Linux-initrd that exposes the block device(s) via the OTG in gadget mode, although MTP might also be an option. That way it will work with whatever hardware the Linux kernel supports rather than baking all of this into the VPU stack.

klslz, 2022-02-24:
I see.  Let me put it my words, just to make sure I understood what you're saying.

You mean a special "maintenance" boot-mode on the NVMe could be an option, that'd put the RPI into gadget mode, while exposing the nvme filesystems.
Similar to what's being done (process wise), with the overlay feature (which also creates a special initrd). Meaning. A new
feature inside raspi-config that would enable gadget mode after a reboot would perhaps be required.

timg236, 2022-02-24:
If it works then msd.elf will be replaced by a boot.img file containing a minimal Linux init-rd that exposes block devices via USB OTG gadget mode. It won't be Raspberry Pi OS or interactive and in the first instance would just replace what the custom msd.elf firmware does i.e. export EMMC. In principle it could export other block devices and since it's just a buildroot image anyone can roll their own.

timg236, 2022-03-10:
@klslz I've uploaded an early release of the rpiboot replacement.  https://github.com/raspberrypi/usbboot/pull/121

The usage is pretty much the same as msd but will will export NVMe, SCSI block devices in addition to EMMC

```
cd mass-storage-gadget
../rpiboot -d .
```

klslz, 2022-03-11:
Just ran a test against my Waveshare board and a Samsung PM991.
It works! Great stuff.

I had to change the clone command though.

`git clone https://github.com/timg236/usbboot.git usbboot-msg`

Thanx a lot for your efforts.

I have a couple more drives and MBs over here. I'll do some more
testing next week.

timg236, 2022-03-11:
> Just ran a test against my Waveshare board and a Samsung PM991. It works! Great stuff.
>
> I had to change the clone command though.
>
> `git clone https://github.com/timg236/usbboot.git usbboot-msg` [...]

Thanks. Hopefully, it can be merged into the main repo next week.

klslz, 2022-03-12:
2nd test:

DFRobot DFR0767 MBO + CM4 - 8GB EMMC

OK!!

klslz, 2022-03-13:
Hi Tim.

How would the eeprom update (recovery) work now??

klslz, 2022-03-14:
So. Final feedback.

I just tested the Eeprom update. It also works.
I also tested a Samsung 980Pro on a PCI-E adapter on the CM4 IO Board. Works great too.

Bottom line.
3 MBOs, 3 media types. All OK.

From my POW the case can be closed as soon as the update is merged.

Thx again.

timg236, 2022-03-14:
@peterharperuk @pelwell ok to merge?

timg236, 2022-03-14:
@klslz Thanks for the test report. This is now merged.

closed 2022-03-14 by timg236