raspberrypi/usbboot #120: Raspberry Pi Imager + rpiboot takes 40 mins to flash 8GB memory

mangelozzi, 2022-03-07:
Using:

- `Raspberry Pi Imager v1.7.1` and 
- `RPIBOOT: build-date Jan 26 2022 version 20220111~130126`

And it takes 40 mins to flash a 8GB CM4 eMMC device (with rpiboot jumper set).

Flashing 12 units takes me a whole day. Is there anyway to speed it up?

JamesH65, 2022-03-07:
There clearly something wrong there, given you are only flashing 8GB. Any unusual setups? Have you tried the Provisioner software as an alternative for multiple device flashing? https://github.com/raspberrypi/cmprovision

mangelozzi, 2022-03-07:
> There clearly something wrong there, given you are only flashing 8GB. Any unusual setups? 

I should mention its 40 mins to write and verify. I don't think I have an unusual setup, heres the setup:
[image]

> Have you tried the Provisioner software as an alternative for multiple device flashing? https://github.com/raspberrypi/cmprovision

Does it offer a speed benefit via ethernet, or just parallelism?

I can use rpiboot to flash multiple targets at the same time. However the problem comes in dev situations, when I would like to revert an image to test the upgrade process. I have to keep waiting 40 mins for every change.

Heres the process I figures out with rpiboot:

- Close or open all instances/windows of `Raspberry Pi Imager` so they add or remove storage targets in the same order.
- Ensure the terminals' `rpiboot` jumper is set.

1. First figure out which USB port number maps to which physical USB port on the computer:
    1. Disconnect all devices
    2. `rpiboot -1` (although `-0` is something to try if desperate)
    3. If you see it spamming lots of lines about send/loading code, that's not the port.
    4. If it connects then you know that port matches the one you tried
    5. Then try again for ports 2-6 until you have mapped which port number to physical USB ports.
2. Now plug in one device into a known USB port number
3. `rpiboot -n` where n a known USB port number
4. Now plug the device into the USB port, it should connect.
5. Leaving the existing devices plugged in, repeat until all targets are mounted.
6. Now open all windows of `Raspberry Pi Imager`, or they should be open from previous attempt.
7. Now start flashing one at a time, before starting the next. Only once `Writing...` has started, start the next one.
8. Turn power off to all units, disconnect `rpiboot` jumper.

JamesH65, 2022-03-08:
Thanks, I' try to find some time to check this out.

timg236, 2022-03-08:
"Does it offer a speed benefit via ethernet, or just parallelism?"

cmprovision downloads the data over using the gbit ethernet and is able to take advantage of the write behind caching when writing to the EMMC without having the overhead of going via USB-2, lots of small transfer and frequent flushes because the host will view it as a removable disk.

In future, rpiboot is likely to be replaced with a Linux init-rd mass storage gadget because it supports arbitrary block devices and is a bit faster than the old VPU code.

JamesH65, 2022-03-08:
I just programmed a CM4/16GB with the latest full Raspberry Pi OS/32bit, using rpiboot and Imager1.7.1 and it took 21 minutes, including verification. Standard CM4IO board. Not sure how large the latest image is though.

timg236, 2022-03-11:
@mangelozzi  The first release of the Linux initrd imager that I mentioned is now available. For me it flashed the normal Raspberry Pi OS in about 5 mins
Instructions here https://github.com/raspberrypi/usbboot/issues/119#issuecomment-1064093364

cmprovision will give the best performance  because it's downloading the data over 1Gbit Ethernet, however, it's a bit more complicated to setup.

timg236, 2022-03-11:
N.B. Raspberry Pi Imager (embedded - https://github.com/raspberrypi/rpi-imager ) can also be loaded via RPIBOOT. This is also quite fast because the network download happens on the CM4. However, it doesn't export the disks via MSD so it depends on what you what to do.

The boot.img can be downloaded from here https://downloads.raspberrypi.org/net_install/ 

At some point I'll write a shell script to usbboot to downloaded the latest boot.img and launch rpiboot to add the embedded imager as an official "rpiboot program"

mangelozzi, 2022-03-15:
> @mangelozzi The first release of the Linux initrd imager that I mentioned is now available. For me it flashed the normal Raspberry Pi OS in about 5 mins Instructions here [#119 (comment)](https://github.com/raspberrypi/usbboot/issues/119#issuecomment-1064093364)
> 
> cmprovision will give the best performance because it's downloading the data over 1Gbit Ethernet, however, it's a bit more complicated to setup.

Wow it is fast, I tried the tool with an official lite image, started it, checked back 10 mins later and it was done! Thank you for this super speedy tool!

I think part of the slowness is my custom image is bigger than the standard images, my Lite (headless) image has grown to about 5GB, even though its only contains a 25Mb git check out and a handful of apt install small programs. 

Before I make an image I have a script that does the following:
```
swapoff then delete the swap file
sudo rm -rf /tmp/*
history -c
sudo apt-get -s clean
Delete chromium data/cache
```

closed 2022-03-15 by mangelozzi