raspberrypi/usbboot #127: [RFC?] Read board revision and serial number

bryceschober, 2022-03-25:
Is it possible to use rpiboot (or enhance it) to get a dump of the connected CM's board revision and serial numbers? This would be really handy to use in production...

timg236, 2022-03-28:
This information is reported in the USB string descriptor from bootcode4.bin but it also reported by the new mass-storage-gadget. Since the descriptor is generated by a Linux script you can put anything there

https://github.com/timg236/cm4-scripts/blob/master/usr/local/bin/configure-gadgets#L33
`
cd mass-storage-gadget
../rpiboot -d .
`

From USB Device Tree Viewer https://www.uwe-sieber.de/
```
             ------ String Descriptor 3 ------
bLength                  : 0x22 (34 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "10000000b29d2da0"
Data (HexDump)           : 22 03 31 00 30 00 30 00 30 00 30 00 30 00 30 00   ".1.0.0.0.0.0.0.
                           30 00 62 00 32 00 39 00 64 00 32 00 64 00 61 00   0.b.2.9.d.2.d.a.
                           30 00                                             0.
```

bootcode4.bin could probably be updated to dump string descriptors e.g. boot the CM4 without a firmware .elf  so that it just sits there and waits
```
mkdir serial
cp msd/bootcode4.bin serial
cd serial
../rpiboot -d .
```
bit busy to look at this right now but hopefully that's enough to hack something in main.c

bryceschober, 2022-03-28:
@timg236 
1. Ok, I now see that I can get the serial number by finding the right node and reading `/sys/bus/usb/devices/*/serial` So what about the board revision?
2. The `rpiboot -vv` output does print a serial number, but not the right one that is available in the USB string descriptor.

bryceschober, 2022-03-28:
@timg236 [This forum post](https://forums.raspberrypi.com/viewtopic.php?p=1168087&sid=62cc5dc1871d4279742b4cb0827b8730#p1168087) seems to indicate that rpiboot used to provide some functionality that it no longer does... at least not the correct serial number for CM4.

timg236, 2022-03-31:
The board revision (and serial number) can be read from /proc/cpuinfo.

I don't think rpiboot has ever automatically extracted these values on 2711 and it might require a patch to bootcode4.bin to make this work nicely. I think it's useful but can't promise when we'll be able to do this.

tdewey-rpi, 2024-11-05:
This can be done using the metadata collection flags.

Set up your config txt as per https://github.com/raspberrypi/rpi-sb-provisioner/blob/b8dfe1550e06f85aabddfb3ed59c4a81cd0e3ef2/service/rpi-sb-provisioner.sh#L417

And then invoke `rpiboot` with the `-j` flag, as per https://github.com/raspberrypi/rpi-sb-provisioner/blob/b8dfe1550e06f85aabddfb3ed59c4a81cd0e3ef2/service/rpi-sb-provisioner.sh#L432

Closing as 'Fixed'. It's pretty useful.