raspberrypi/usbboot #131: Secure boot error - Firmware not found

Vijay-P1999, 2022-04-19:
Hey, Tried secure-boot for RPI Compute module 4. Followed all the steps mentioned in this [Repo](https://github.com/raspberrypi/usbboot). But ended up with an error. 

[image]

Is the documentation provided in this repo are accurate and up-to-date?

> Clone the Raspberry Pi OS boot files
> Copy the contents of /boot to a local directory called secure-boot-files

**Question 1: Which `/boot` content should i copy into `secure-boot-files` for generating img? `/Firmware/boot` or `/boot` partition in Raspberry pi OS?**

>Copy the secure boot image to the boot partition on the Raspberry Pi.
>Copy boot.img and boot.sig to the chosen boot filesystem. Secure boot images can be loaded from any of the normal boot devices (e.g. SD, USB, Network).

**Question 2: What should be the final content of `/boot`? Should it contain only these files?**

` boot.img`, `boot.sig`

## Already tried:
Tried flashing both 32 & 64 bit OS to the CM4 and generated `boot.img` from respective `boot` partition. Signed the img with private.pem and copied it to CM4. But still, the same error. Couldn't find any relevant issues/fix related to this. 

## Environment:
Ubuntu 20.04.3 LTS
Waveshare CM4-IO-Base-A with Raspberry Pi compute module 4

peterharperuk, 2022-04-19:
signed boot will only require boot.img and boot.sig. The img file contains the files needed. It looks like you haven't enabled signed boot SIGNED_BOOT=1? See https://github.com/raspberrypi/usbboot/blob/master/secure-boot-recovery/README.md

Vijay-P1999, 2022-04-20:
Thanks, I didn't enable it because it was mentioned optional in the document.

> Optional. Customize the EEPROM config.

I tried booting cm4 after enabling `SIGNED_BOOT=1` in `boot.conf` . I get this.

[image]

And also which steps should i follow for secure-boot in cm4 - [usbboot](https://github.com/raspberrypi/usbboot#secure-boot) or [secure-boot-recovery](https://github.com/raspberrypi/usbboot/blob/master/secure-boot-recovery/README.md#raspberry-pi-4---secure-boot) ?

timg236, 2022-04-20:
You need to use secure-boot-recovery to flash the an EEPROM with secure-boot locked to your private key.

N.B. I'm assuming that you have already decided how your OS image will work i.e. initramfs etc

Vijay-P1999, 2022-04-20:
Thanks @timg236 I have no idea about how OS image will internally work. I'm just following this repo for secure-boot in CM4. 

> You need to use secure-boot-recovery to flash the an EEPROM with secure-boot locked to your private key.

So i guess this is next step after booting with signed `boot.img`. But do you have any idea why i'm getting this error(Refer Image)?

timg236, 2022-04-20:
For an OS image I'd recommend starting with the secure-boot-example which is self contained busybox image using an initframfs for the boot partition. To be secure the Kernel + m odules need to be loaded from a file-system that is verified by the signature and the easiest way to do that is have all of that in an initframfs that is loaded from boot.img
https://github.com/raspberrypi/usbboot/tree/master/secure-boot-example

I suspect that the EEPROM config file doesn't have SIGNED_BOOT=1 so it will be looking for start.elf instead OR it does contain that and the SD-card is missing boot.img or boot.sig

For debugging secure-boot it's worth having a USB serial adapter because there's a lot more debug information available if you specify BOOT_UART=1 in config.txt

Vijay-P1999, 2022-04-20:
> I suspect that the EEPROM config file doesn't have SIGNED_BOOT=1 so it will be looking for start.elf instead OR it does contain that and the SD-card is missing boot.img or boot.sig

I enabled SIGNED_BOOT=1 and i have `boot.img` & `boot.sig` in `/boot` of RPI. But still the same error persists. This is what confusing me.

Vijay-P1999, 2022-04-20:
@timg236 When i was trying various options of secure-boot in CM4,Suddenly now I'm having problem with running `sudo ./rpiboot` 
At this point, it get stuck
```
user1@my-pc:$ sudo ./rpiboot
RPIBOOT: build-date Apr 19 2022 version 20220315~121405 445356e1  
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Successful read 4 bytes 
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Second stage boot server
```

I couldn't re-flash any OS into the CM4 also. Is the board bricked? Any suggestions?

timg236, 2022-04-20:
Running rpiboot with  no arguments will put the device into mass-storage mode that's the normal behaviour. 

You can then copy boot.img / boot.sig to the sd-card (CM4 lite) or EMMC

timg236, 2022-04-26:
Closing since this does not appear to be a bug
https://forums.raspberrypi.com/viewtopic.php?p=1997280#p1997280

closed 2022-04-26 by timg236