raspberrypi/usbboot #135: CM4 hangs

mikev1963, 2022-05-01:
I am trying to load a new os, but when I run the following command it Hangs,  Ran for 1 day and I only see the following:

sudo ./rpiboot
RPIBOOT: build-date May  1, 2022 version 20220427-095022 54a0395d
Waiting for BMC2835/6/7/2711...
Loaded embedded: bincode4.bin
Sending bootcode.bin
Successfully read 4 bytes
Waiting for BCM2835/6/7/2711....     **<- hangs here for over 1 day,**

in /var/log/syslog:

usb disconnect errors

Vishwasrao1, 2022-05-02:
I am also facing same issue on MacOs 

- I also tried to downgrade libusb version to previous one that I was using with old usbboot., however it won't work.  

- I tried changing usb cables so it won't work.

- Also I  tried to restore previous version of usbboot however it shows, rpiboot command not found.

mikev1963, 2022-05-02:
> I have resolved this issue, You need to update update the SPI EEPROM bootloader on a Compute Module 4. Follow these steps
> cd recovery ./update-pieeprom.sh ../rpiboot -d .
>
> then you can use rpiboot to flash your device.

This did not work for me😂

mikev1963, 2022-05-02:
I got it to work.  Here is what I did.

1. powered off RPI CM4.
2. Ran the steps above from Vishwarao1
3. Powered on device.

Worked like a champ.  Crazy.

closed 2022-05-02 by mikev1963