raspberrypi/usbboot #138: Secure Boot on Buster

jwainwright87, 2022-05-11:
Hi all,

My use case means that I have to use Buster for the time being, and I wanted to confirm if secure boot is supported under Buster?

I have achieved this using the latest Buster OS version via RPI Imager.

Is this officially supported, or am I in dangerous territory?

I am using CM4.

Thanks,
Jamie

timg236, 2022-05-11:
Secure-boot is just a mechanism for loading a signed ramdisk containing GPU firmware, kernel, initrd etc so there are no limitations on the host OS. The Buster start4.elf firmware supports secure-boot but it might be best to grab the latest firmware from Bullseye.

You'll need to create an initrd with modules overlays etc for secure-boot so it may be best to experiment with a simple busybox ramdisk before attempting to do this on a full OS.
e.g. https://github.com/raspberrypi/usbboot/tree/master/secure-boot-example

jwainwright87, 2022-05-11:
Great, thanks for the prompt response

closed 2022-05-11 by jwainwright87

jwainwright87, 2022-05-11:
> The Buster start4.elf firmware supports secure-boot but it might be best to grab the latest firmware from Bullseye.

@timg236 Just to clarify, do I do this by grabbing start4.elf from the latest release and replacing it?

timg236, 2022-05-11:
That would work but if you are scripting this it’s better to grab start4.dat and fixup4.dat from the stable branch of the firmware repo https://github.com/raspberrypi/firmware