raspberrypi/usbboot #139: How to check if ./usbboot/rpiboot has already been ran?

iris2312, 2022-05-12:
I am using a Raspberry Pi and rpi-imager to flash images to raspberry pi Compute Modules (CM3s). I notice that it hangs if I run usbboot/rpiboot  more than once, without unmounting a CM3. How can I handle this more gracefully by checking (using code) if it was already ran once?

klslz, 2023-01-20:
I was just about to write a trouble ticket about this.

"rpiboot hangs if CM4 is already booted"

1. rpiboot should be able to recognize that "already booted" state and bail out immediately or
2. the todays infinite  "waiting" period should run into a timeout
3. an ERROR message should be issued that the CM4 is already booted and needs to be repowered to run rpiboot

otherwise if you try (bash-)scripting the process, you need to do a lot of extra work to cope with the issue.

klslz, 2023-01-20:
I was just about to write a trouble ticket about this. To me it's an error.

my reference commit-level: https://github.com/raspberrypi/usbboot/commit/864863bc1944553f5a4a7dae5df9ea894afdf513 (Jan-20-2023)

```rpiboot hangs if CM4 is already booted```

1. rpiboot should be able to recognize the "already booted" state and bail out immediately or
2. the todays infinite "waiting" period should run into a timeout
3. an ERROR message should be issued that the CM4 is already booted and needs to be "repowered" to run rpiboot
    otherwise if you try (bash-)scripting the process, you need to do a lot of extra work to cope with the issue.

I do also need to "repower" the CM4 if I e.g. first boot&mount  and after that I'd like to run  a firmware upgrade.  
The whole repowering thing is not nice. It would be nice if a hardware reset could be issued by the tool. This way
the whole maintenance process could be scripted without the need for plugging cables all the time.

I just realized that's an open issue since 8 month. Let's hope someone ( @timg236 ) finally picks it up. Tim are you watching?

pelwell, 2023-01-20:
This is not possible, except by adding a timeout or detection of a specific USB gadget that all images would have to run. See #179.

closed 2023-01-20 by pelwell