raspberrypi/usbboot #143: secure-boot is getting cleared over image a recovery image

Narashiman, 2022-05-27:
Hardware Used :  CM4 (8GB eMMC, 1GB RAM) with CMIO board

Steps done : 
git pull of https://github.com/raspberrypi/usbboot
generated a RSA-key as mentioned in README file and generated a boot.sig file and flashed it. Able to read the eMMC as storage only if I boot with signed boot file in folder [secure-boot-recovery](https://github.com/raspberrypi/usbboot/blob/master/secure-boot-recovery/README.md), [secure-boot-msd](https://github.com/raspberrypi/usbboot/blob/master/secure-boot-msd/README.md) and [secure-boot-example](https://github.com/raspberrypi/usbboot/blob/master/secure-boot-example/README.md). Not able to read eMMC as storage via default or other key-pair folder. 
Already set program_pubkey=1 in config.txt file.

Issue :  
When i try to do ./rpiboot -d recovery/ . It overrides all and was able to read the eMMC as storage once again.

pelwell, 2022-05-27:
Where does it say that secure boot prevents accessing the eMMC using rpiboot? The "secure"ness stops the unit from booting an image which has been tampered with, but that is all.

Narashiman, 2022-05-27:
Yes. Isn't the boot file in recovery folder invalid against the "secure"ness check since it contains a different or no RSA-key.

peterharperuk, 2022-05-27:
Have you enabled secure boot (Locked secure-boot mode)? Until that's done (please don't do it until you're ready!) I believe there's nothing to stop you going back to the old bootloader?

Narashiman, 2022-05-27:
As mentioned, I have Already set program_pubkey=1 in config.txt file
Is there anything else to be done. I am all ready with my OS image and is found to be booting and working fine after signing.

peterharperuk, 2022-05-27:
Sorry missed that comment. Do you have a uart? After you've run "rpiboot -d recovery" what happens? Does it boot? In which case it doesn't sound like enabling secure boot worked. Can you capture uart output?

timg236, 2022-06-06:
In addition to capturing the host + UART output for `cd secure-boot-recovery; ../rpiboot -d ` Please can you run capture the output of the following commands to clarify whether the OTP was updated for secure-boot mode.

`vcgencmd otp_dump`
and
`hexdump -C /proc/device-tree/chosen/bootloader/signed`

timg236, 2022-07-08:
Closing due to inactivity. Please re-open if you can provide the logs according to the last comment

closed 2022-07-08 by timg236