raspberrypi/usbboot #144: [Question] Where is the source located of `msd/bootcode.bin` & `msd/start*.elf`?

WeeJeWel, 2022-05-30:
I'd like to view an example on how to compile my own bootable firmware (specifically to try flashing a CM4 over WebUSB for fun).

Is this source available anywhere?

pelwell, 2022-05-30:
The VPU firmware is closed-source. The recommended method for a task such as the one you describe is to create a Buildroot Linux system with an initrd that does the job.

timg236, 2022-06-10:
Take a look at the msd gadget for an example kernel and initrd. The ARM stage loading does not require Linux so you can replace the kernel with bare metal
ARM code

closed 2022-06-10 by timg236

cleverca22, 2022-06-25:
> specifically to try flashing a CM4 over WebUSB for fun

already implemented that!
https://github.com/librerpi/rpi-tools/tree/master/webusbboot

WeeJeWel, 2022-06-25:
> > specifically to try flashing a CM4 over WebUSB for fun
> 
> already implemented that! https://github.com/librerpi/rpi-tools/tree/master/webusbboot

Yes, that’s nice, but it mounts as MSD which then isn’t available anymore from a browser.