raspberrypi/usbboot #145: Is it possible to use Uboot from inside a secure signed boot.img?

3dlabsio, 2022-06-10:
I've tried utilizing uboot within a boot.img on a cm4 configured for secure signed boot images without success. I've also validated that it boots with a vanilla set of files in the boot.img.

I'll have to dig into the cause of the failure more, but wondering if anyone out there has successfully configured this.

timg236, 2022-06-10:
This is really a question for the forums since we don't support u-boot.

The GPU firmware loads a generic ARM blob + stubs and initramfs if specified. Since you are inserting an  extra boot stage the initramfs within  boot.img would need to contain all the resources require by u-boot since boot.img is discarded as soon as the kernel is loaded.

closed 2022-06-10 by timg236

timg236, 2022-06-10:
Edit. The ramdisk is loaded at 512MB if you really want to patch uboot to read this. Although I’d recommend using a proper initrd like the Linux kernel

embed123linux, 2024-02-13:
Hi @3dlabsio did you manage to u-boot within `boot.img`? I included `boot.scr` and `uImage` file inside `boot.img` but this does not seem to work.