raspberrypi/usbboot #146: libusb.h: No such file or directory

biswaKL, 2022-06-17:
Hi Team,
I am using Ubuntu 22.04,
I have INstalled libusb as well, according to README
But i am still facing below error:

```
biswa@biswa-laptop:~/Softwares/usbboot$ make
cc -Wall -Wextra -g -o bin2c bin2c.c
./bin2c msd/bootcode.bin msd/bootcode.h
./bin2c msd/start.elf msd/start.h
./bin2c msd/bootcode4.bin msd/bootcode4.h
./bin2c msd/start4.elf msd/start4.h
cc -Wall -Wextra -g -o rpiboot main.c `pkg-config --cflags --libs libusb-1.0` -DGIT_VER="\"726256d9\"" -DPKG_VER="\"20220504~214218\""
/bin/sh: 1: pkg-config: not found
main.c:1:10: fatal error: libusb.h: No such file or directory
    1 | #include <libusb.h>
      |          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:4: rpiboot] Error 1

```

pelwell, 2022-06-17:
Without trying it myself, I think this is the significant line:
```
/bin/sh: 1: pkg-config: not found
```
Have you tried:
```
$ sudo apt-get install -y pkg-config
```
?

biswaKL, 2022-06-17:
Thant worked, thanks.

closed 2022-06-17 by biswaKL

pelwell, 2022-06-17:
I've added pkg-config installation to the README.