raspberrypi/usbboot #147: Guidance required : generate boot.img and boot.sign using yocto project for secure boot

Vishwasrao1, 2022-06-18:
As of now , I was manually making boot.img, signing it using usbboot tool for my custom yocto based image. 

I would like to incorporate above step in yocto toolchain. so that I will get final signed image from yocto itself. The hardware is rpi CM4. Once I have achieved this I would like to enhance this and generate complete chain of trust including rootfs with help of yocto project.

I was not sure how to start with this, I was wondering if someone from the community has already given it a try or know how to proceed with this. If anyone can guide me or would like to contribute that will be really helpful.

pelwell, 2022-06-19:
Perhaps it would help if you start by sharing your manual flow - the additional steps required to make your signed image. They might be obvious to you, but it helps to have something concrete to work with.

Vishwasrao1, 2022-06-20:
@pelwell Thank you for your reply. The steps are. 

1. Generate custom linux image using yocto.
2. Flash the image to compute module 4 using usbboot/rpiboot.
3. Check content of the boot partition. copy the content to a folder secure boot files.
4. Generate a boot image using make-boot-image in usbboot.
5. Sign the boot image using rpi-eeprom-digest of usbboot.
6. copy paste the boot.img and boot.sign in boot partition of cm4
7. Now the cm4 boots in secure boot mode using the boot.img and boot.sign

What I want to achieve:
1. Generate boot.img and boot.sign in yocto itself so that I don't have to do steps 2,3,4,5,6.
2. The yocto project will generate boot.img , sign the image and place it in boot partition of my linux OS. 

Let me know if you have any idea or  if you have already achieved this.

pelwell, 2022-06-20:
This is something that we are interested in, primarily for buildroot but also Yocto, but it isn't a high priority (just to set your expectations). The first hurdle is likely to be the creation of a suitable initramfs, the hard part being giving the user a reasonable degree of control over what goes into it without it becoming just a huge list of files.

timg236, 2022-06-20:
In buildroot you'd typically add something to post-build.sh to collect the necessary (firmware, config, cmdline, initrd) and run make-boot-image. That would create a boot.img file which is then included by genimage.

Presumably, Yocto has equivalent post-build hooks, but I don't know what they would be.

Vishwasrao1, 2022-06-23:
@pelwell Thank you for clarification. 
@timg236 Yes, I am going to look for any such post build script. This has set my direction. I will keep this issue open for others to reply, I will also post updates if I achieve something regarding this.

timg236, 2022-10-25:
There is now an example buildroot board/config for building and signing a boot.img
https://github.com/raspberrypi/buildroot/blob/raspberrypi-signed-boot/README.md

closed 2024-01-08 by timg236

BuriKizilkaya, 2025-03-06:
@Vishwasrao1 

Have you found a solution? I can not find any related documentation how to create boot.img from yocto for the raspberry pi

timg236, 2025-03-06:
Please don't wake up old bugs unless you are the OP.  The Raspberry Pi forums are the place to ask general support questions
https://forums.raspberrypi.com/