raspberrypi/usbboot #151: rpiboot.exe not running, Windows returns error code 0xc000007b

pilcru, 2022-07-18:
Since the last commit to `rpiboot_setup.exe` (a490517d0b1fca5b369080c3d9ca43d3f59786c9), installing rpiboot on Windows 10 and trying to open `rpiboot.exe` throws this error:

---------------------------
rpiboot.exe - Application Error
---------------------------
The application was unable to start correctly (0xc000007b). Click OK to close the application. 
---------------------------
OK   
---------------------------

Using the exe from the previous commit (0aad0470420d8a93cce41ae32a210f4a14703f99), rpiboot.exe works fine.

I'm using the same procedure in both cases, and tried installing and running as administrator.

timg236, 2022-07-18:
Please can you try f4273e9201f674392f790cae9a109662d62291ec ?

pilcru, 2022-07-18:
That installs and runs fine, however I do not have a board on hand to test it all the way to mounting the eMMC.

```
RPIBOOT: build-date Jul 18 2022 version 20220718~085937 5a25e04b
Waiting for BCM2835/6/7/2711...
```

timg236, 2022-07-19:
Thanks for testing. I think the previous .exe was corrupted. If it's running then the installed files should be fine.

pilcru, 2022-07-22:
Just wanted to confirm I tested connecting to a CM4 today from Windows 10, using version f4273e9, and it's all working fine. Thanks for the quick response and fix!

closed 2022-07-22 by pilcru