raspberrypi/usbboot #152: mass-storage-gadget: No mass-storage on CM4 / CM4S

nbuchwitz, 2022-08-15:
I tried to boot a CM4 (in official IO board) and a CM4S (in official IO board and also in our RevPi), but in both cases I was unable to get a mass storage device on my host system.

After some debugging I noticed that the file `dwc2-overlay.dtb` is missing in the `boot.img`. With this file added and also adapting `dtoverlay=dwc2,dr_mode=otg` to `dtoverlay=dwc2,dr_mode=otg` in `config.txt` both devices work as expected.

This hasn't been tested on a Raspberry Pi 4 or 400, but I expect the same behavior.

If the above solution sounds sane to you, I'm happy to provide a PR with things fixed.

pelwell, 2022-08-15:
The expected overlay name is `dwc2.dtbo` - which distro and kernel version are you running?

timg236, 2022-08-15:
It seems to work for me

timg236, 2022-08-15:
dwc-otg.dtbo and dwc2.dtbo are also in the boot.img overlays directory in this file https://github.com/raspberrypi/usbboot/blob/master/mass-storage-gadget/boot.img

I'm not aware of a dwc2-overlay.dtb file. I don't think it's a thing

nbuchwitz, 2022-08-15:
Never mind the dwc2 thing, looks like a mess-up during debugging. The main things seems to be the `dr_mode=peripheral`. I just verified with `mass-storage-gadget` from this repository as following:

CM4 on CM4IO: works
CM4S on CM4IO: doesn't work (boots, but never exposes mass storage)

Kernel log from CM4S:

```
[...]
[    0.840817] of_cfs_init
[    0.840918] of_cfs_init: OK
[    0.841839] uart-pl011 fe201000.serial: no DMA platform data
[    0.869056] mmc0: SDHCI controller on fe340000.mmc [fe340000.mmc] using ADMA
[    0.873596] Freeing unused kernel memory: 2048K
[    0.873939] Run /init as init process
[    0.873948]   with arguments:
[    0.873957]     /init
[    0.873965]   with environment:
[    0.873973]     HOME=/
[    0.873981]     TERM=linux
[    1.002905] mmc0: new DDR MMC card at address 0001
[    1.003761] mmcblk0: mmc0:0001 BJTD4R 29.1 GiB
[    1.004172] mmcblk0boot0: mmc0:0001 BJTD4R partition 1 4.00 MiB
[    1.004581] mmcblk0boot1: mmc0:0001 BJTD4R partition 2 4.00 MiB
[    1.004834] mmcblk0rpmb: mmc0:0001 BJTD4R partition 3 4.00 MiB, chardev (245:0)
[    1.005717]  mmcblk0: p1 p2
[    1.019822] udevd[111]: starting version 3.2.11
[    1.020671] random: udevd: uninitialized urandom read (16 bytes read)
[    1.020877] random: udevd: uninitialized urandom read (16 bytes read)
[    1.020974] random: udevd: uninitialized urandom read (16 bytes read)
[    1.032460] udevd[112]: starting eudev-3.2.11
[    1.157664] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    1.158999] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[    1.159021] [vc_sm_connected_init]: start
[    1.160810] [vc_sm_connected_init]: installed successfully
[    1.189546] dwc2 fe980000.usb: supply vusb_d not found, using dummy regulator
[    1.189772] dwc2 fe980000.usb: supply vusb_a not found, using dummy regulator
[    1.407069] dwc2 fe980000.usb: DWC OTG Controller
[    1.407109] dwc2 fe980000.usb: new USB bus registered, assigned bus number 1
[    1.407152] dwc2 fe980000.usb: irq 37, io mem 0xfe980000
[    1.407405] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.10
[    1.407420] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.407433] usb usb1: Product: DWC OTG Controller
[    1.407444] usb usb1: Manufacturer: Linux 5.10.95-v7l dwc2_hsotg
[    1.407455] usb usb1: SerialNumber: fe980000.usb
[    1.408114] hub 1-0:1.0: USB hub found
[    1.408164] hub 1-0:1.0: 1 port detected
[    1.579004] Mass Storage Function, version: 2009/09/11
[    1.579023] LUN: removable file: (no medium)
[   31.196379] cam-dummy-reg: disabling
[  206.956423] random: crng init done
[  206.956436] random: 3 urandom warning(s) missed due to ratelimiting
```

My test systems where a Raspberry 3B+ running latest Buster image (5.10.120) and my workstation running Debian Bullseye (kernel 5.10.127). If I modify the `config.txt` it works on CM4 and CM4S as expected.

nbuchwitz, 2022-08-15:
Would make sense, as it is the only USB. Is there anything which speaks against having `dr_mode=peripheral` as the default?

timg236, 2022-08-15:
I think that would be ok

nbuchwitz, 2022-08-15:
Fixed with f765585dcd438117a7d67ce39060d24391c69b29

closed 2022-08-15 by nbuchwitz

timg236, 2022-08-15:
Thanks for confirming and the debugging :)