raspberrypi/usbboot #153: Unable to reach stage 2. Brand new CM4

c157fl, 2022-08-15:
**Model:** CM4102008

**Objective:**:
- Flash EMMC

**Issue:**:
- Running usbboot does not lead to the storage appearing.

**Available carrier boards:** 
- dfrobot router board
- pitray mini
- pibox
- mcuzone tiny cs*2

**Available systems:** 
- W11
- Pi4B(64OS)

**Available cables:** USB-A to USB-C, USB-C to USB-C

Usbboot behaviour on both Pi4B and W11:
```
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
```

Different combinations of cables and carrier boards were tried to no avail.

When at this stage the ACT green LED blinks 8 times in short pulses.
According to the documentation this would imply faulty SDRAM.

Is there anything I can do to recover it or is my only option to send it back to the seller?

Thank you in advance

timg236, 2022-08-15:
The logs suggest this is an old version of usbboot. It might be worth updating that first

c157fl, 2022-08-15:
Thank you.

On my Pi4B I had done the:
```
sudo apt install git libusb-1.0-0-dev pkg-config
git clone --depth=1 https://github.com/raspberrypi/usbboot
cd usbboot
make
sudo ./rpiboot
```

If I do that in the Pi4B with the verbose flag I have:
```
sudo ../rpiboot -d . -v
RPIBOOT: build-date Aug 13 2022 version 20220804~165403 3e7df2a1
Boot directory '.'
Loading: ./bootcode4.bin
Waiting for BCM2835/6/7/2711...
Loading: ./bootcode4.bin
Device located successfully
Initialised device correctly
Found serial number 3
Sending bootcode.bin
libusb_bulk_transfer sent 24 bytes; returned 0
Writing 87944 bytes
libusb_bulk_transfer sent 87944 bytes; returned 0
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
```

timg236, 2022-08-15:
It might be worth trying the recovery and then mass-storage-gadget instead of msd/

c157fl, 2022-08-15:
Thank you @timg236.

Since doing the `sudo rpiboot -d recovery` hangs (I'm assuming this is flashing the EEPROM) hangs when doing `Waiting for BCM2835/6/7/2711` for the second stage, I Interrupted and attempted at doing `sudo rpibood -d mass-storage-gadget`.

Unfortunately I did not have success. It does not get detected.

If I then reset the device, it gets detected and does stage 1 but then hangs again:
```
RPIBOOT: build-date Aug 13 2022 version 20220804~165403 3e7df2a1
Boot directory 'mass-storage-gadget'
Loading: mass-storage-gadget/bootcode4.bin
Waiting for BCM2835/6/7/2711...
Loading: mass-storage-gadget/bootcode4.bin
Device located successfully
Initialised device correctly
Found serial number 3
Sending bootcode.bin
libusb_bulk_transfer sent 24 bytes; returned 0
Writing 106008 bytes
libusb_bulk_transfer sent 106008 bytes; returned 0
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
```

timg236, 2022-08-15:
This sounds like a hardware issue. You could try older versions of usbboot

c157fl, 2022-08-15:
Did:
```
git checkout XXXX
make 
cd recovery
../rpiboot -d . -v
```
to versions:

| cc72bc01ac | May 4 | Hangs |
| 42fe88dac5 | April 26 | Hangs |
| 3ef29d3ab4f5369b919ddf8742df2ec5cc9a05b5 |  Feb 8 |  Hangs |
| d49a1a2d72afcfc3e5fdaf2218975622b88bf5e1 |  Jan 27 |  Hangs |
| 0eb1b63306e8a4980fc6993d10a28d434277efab |  Jan 27 |  Hangs |
| 97f618fccce4cc7bacaa8bb883da63f06f9854df |  Jan 11 |  Failed to write correct length, returned -7 |
| 515b972ff0f864daeeecb69f9dc6105d2129f0cc |  Dec 14 |  Failed to write correct length, returned -7 |

Unfortunately with no success. 
Keep being _stuck at stage 2_  with the _blinking 8 times LED_.

timg236, 2022-08-15:
Oh dear, that sounds bad, possibly needs an RMA. Do you have another CM4 to cross check it against the carrier boards? Although, given that the DDR voltages comes from the on-module PMIC it's unlikely to be an issue with the carrier board.

c157fl, 2022-08-15:
The carrier boards are good. This CM4 was going to replace a beefier one currently deployed DFRobot's router carrier board.
I talked to the seller but wanted to exhaust all the options. This one took almost a week to arrive (Germany->Portugal) and now sending it back and getting an answer will probably take double of that.
Thank you for help @timg236. I'll be continuing the RMA process then.

closed 2022-08-15 by c157fl

Sandi2211, 2022-11-21:
> The carrier boards are good. This CM4 was going to replace a beefier one currently deployed DFRobot's router carrier board. I talked to the seller but wanted to exhaust all the options. This one [...]

What was the outcome of this? I also have two brand new CM4, one of them is working as it should, the other one like you described.
Also tried several combinations of cables, 4 Operating Systems, Carrier Boards.
I'll need to send it back as well I guess, but there are no new ones in stock so I thought I'll ask here at first. Thanks!

c157fl, 2022-11-23:
> What was the outcome of this? I also have two brand new CM4, one of them is working as it should, the other one like you described. Also tried several combinations of cables, 4 Operating Systems, [...]

Got my refund, even though they did not refund the return costs as previously stated. 
The replacement unit I got while handling the RMA worked directly. No issues like with the former.

Took more than a month of the unit being ~~forgotten somewhere~~ "inspected" and me being annoyed at them for a decision to come through.