raspberrypi/usbboot #155: Add tag on release

bcecchinato, 2022-08-17:
Hello !

Is it possible to tag all releases ? This would allow to clone the tag with depth=1 and not the entire history :)

Thanks in advance.

pelwell, 2022-08-17:
What do you mean by "releases"? A tag is just a label on a commit - commit hashes can be used in just the same way if you know what you want.

bcecchinato, 2022-08-17:
Not exactly, a tag is also a ref in git, which means you can clone like this `git clone --depth=1 your_tag https://github.com/raspberrypi/usbboot`. This will reduce the clone to 1 commit and makes it very fast.

It is not possible to clone based on a sha1 without cloning the whole tree (and thus makes the clone way more longer), except if there is ref attached to the sha1 (like a tag or a branch).

By release I mean every commit name "release XXXX" for instance [20220815~145439 release](https://github.com/raspberrypi/usbboot/commit/1e651a88f0313a5ca50b12a9992dbce8930fe200).

timg236, 2022-08-17:
Realistically no, I'm not going to remember to do this

closed 2022-08-17 by timg236

pelwell, 2022-08-17:
Interesting - I'd not come across the "only named branches/tags" limitation of git clone before. It's apparently a security feature, to protect against old (potentially dangerous or embarassing) commits from being downloaded after they've been removed from a repo but not yet pruned.

bcecchinato, 2022-08-17:
Yep :) I know that other repos like https://github.com/raspberrypi/linux and https://github.com/raspberrypi/firmware are using tags which is very pratical to avoid this problem.

Anyway, that's your call. I know it can be done in 1 click on github (or one push with `git tag XXX + git push origin XXX` Too bad this is not possible to "suggest" as a PR or something :(

jcgruenhage, 2022-08-24:
I'd like to weigh in on this: This is not necessarily about remembering. If you forget to tag it in git, fine, so be it, we'll remember for you and remind you when it happens. For users and distributions alike, it'd be very nice to have tags available, so that packaging this software becomes feasible. Having to keep those tags of which version number refers to which commit hash in each distribution is clearly more work compared to keeping it in one place, upstream.

uffejakobsen, 2022-08-24:
I agree !!!