raspberrypi/usbboot #156: The `-p` option is not available in the Windows executable.

JinShil, 2022-09-01:
Please compare the following usage output on both Linux and Windows...

Output on Linux:
```
pi@raspberrypi:~ $ rpiboot --help
Usage: rpiboot
   or: rpiboot -d [directory]
Boot a Raspberry Pi in device mode either directly into a mass storage device
or provide a set of boot files in a directory from which to boot.  This can
then contain a initramfs to boot through to linux kernel

To flash the default bootloader EEPROM image on Compute Module 4 run
rpiboot -d recovery

For more information about the bootloader EEPROM please see:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

rpiboot                  : Boot the device into mass storage device
rpiboot -d [directory]   : Boot the device using the boot files in 'directory'
Further options:
        -l               : Loop forever
        -o               : Use files from overlay subdirectory if they exist (when using a custom directory)
                           USB Path (1-1.3.2 for example) is shown in verbose mode.
                           (bootcode.bin is always preloaded from the base directory)
        -m delay         : Microseconds delay between checking for new devices (default 500)
        -v               : Verbose
        -s               : Signed using bootsig.bin
        -0/1/2/3/4/5/6   : Only look for CMs attached to USB port number 0-6
        -p [pathname]    : Only look for CM with USB pathname
        -h               : This help
```

Output on Windows:
```
PS C:\Program Files (x86)\Raspberry Pi> .\rpiboot.exe --help
Usage: rpiboot
   or: rpiboot -d [directory]
Boot a Raspberry Pi in device mode either directly into a mass storage device
or provide a set of boot files in a directory from which to boot.  This can
then contain a initramfs to boot through to linux kernel

To flash the default bootloader EEPROM image on Compute Module 4 run
rpiboot -d recovery

For more information about the bootloader EEPROM please see:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

rpiboot                  : Boot the device into mass storage device
rpiboot -d [directory]   : Boot the device using the boot files in 'directory'
Further options:
        -l               : Loop forever
        -o               : Use files from overlay subdirectory if they exist (when using a custom directory)
                           USB Path (1-1.3.2 for example) is shown in verbose mode.
                           (bootcode.bin is always preloaded from the base directory)
        -m delay         : Microseconds delay between checking for new devices (default 500)
        -v               : Verbose
        -s               : Signed using bootsig.bin
        -0/1/2/3/4/5/6   : Only look for CMs attached to USB port number 0-6
        -h               : This help
```

Notice the lack of the `-p` (USB path) option on Windows.  

Is it possible to add this option to the Windows executable?

timg236, 2022-09-01:
It is available, you just need to build it from source. The Windows .exe installer is only updated periodically

closed 2022-09-01 by timg236

JinShil, 2022-09-01:
The `-p` option was added in October of 2020.  See https://github.com/raspberrypi/usbboot/commit/57d692145dda563e40da293efb5cbb415d563c33

The Windows executables were updated in July of this year.  See https://github.com/raspberrypi/usbboot/commit/f4273e9201f674392f790cae9a109662d62291ec#diff-d94917367dfd51caa5452cc16635d700713875420a94e2cc1b6b45ec84217543

Why doesn't the latest Windows executable contain the `-p` option?