raspberrypi/usbboot #157: Instructions to compile insufficient for Fedora 36

altometer, 2022-09-02:
# Issue

| Key        | Value                                   |
| ---------- | --------------------------------------- |
| OS         | Fedora Linux 36 (KDE Plasma) x86_64     |
| Host       | Laptop (12th Gen Intel Core) A4         |
| Kernel     | 5.18.19-200.fc36.x86_64                 |
| Shell      | zsh 5.8.1                               |
| Resolution | 2256x1504, 3440x1440                    |
| DE         | Plasma 5.25.4                           |
| WM         | KWin                                    |
| Terminal   | konsole                                 |
| CPU        | 12th Gen Intel i5-1240P (16) @ 4.400GHz |
| GPU        | Intel Alder Lake-P                      |
| Memory     | 31806MiB                                |

1. `$ dnf install git libusb-1.0-0-dev pkg-config`
2 . `$ git clone --depth=1 https://github.com/raspberrypi/usbboot`
3. `$ cd usbboot; make`

> Output
```bash
cc -Wall -Wextra -g -o rpiboot main.c `pkg-config --cflags --libs libusb-1.0` -DGIT_VER="\"1e651a88\"" -DPKG_VER="\"20220815~145439\""
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libusb-1.0', required by 'virtual:world', not found
main.c:1:10: fatal error: libusb.h: No such file or directory
    1 | #include <libusb.h>
      |          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:4: rpiboot] Error 1
```

# Attempted Fixes

I tried installing `pkgconf` and `ncurses-compat-libs` through dnf.

pelwell, 2022-09-02:
We can't offer specific Fedora support, but you could try installing pkgconf separately, before libusb-1.0-0-dev.

altometer, 2022-09-03:
Thanks! Yep, figured this was user error on my part.

# Working fix

```bash
rm -rfv "<your path here>" # /home/user/usbboot in my case
sudo dnf install libusb1-devel
git clone --depth=1 https://github.com/raspberrypi/usbboot
cd usbboot
make
```

👍

closed 2022-09-03 by altometer