raspberrypi/usbboot #159: Continously loading bootcode4.bin

JannisLuechtefeld, 2022-09-15:
I am facing some strange behaviour for CM4:

```
PIBOOT: build-date Jul 18 2022 version 20220718~085937 5a25e04b

Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode4.bin
Loading embedded: bootcode4.bin
Loading embedded: bootcode4.bin
Loading embedded: bootcode4.bin
Loading embedded: bootcode4.bin
Loading embedded: bootcode4.bin
...
```

This happens to different CM4 on different boards. The boards most of the time do work with a different compute module. Sometimes switching to a different computer also works.

timg236, 2022-09-15:
Seems to be a duplicate of https://github.com/raspberrypi/usbboot/issues/36

trimixtimo, 2022-09-16:
I am facing the same problem from time to time. In comparison to #36 I do not get any error code. It´s just loading the bootcode4.bin forever. Sometimes it works to use another IO-board, sometimes I just use another CM4 and it works. 

The hint that the USB port is not properly controlled by Windows seems conclusive, but the IO-board is powered externally and should therefore not draw any significant current from the usb port.

timg236, 2022-09-16:
This will just be an artifact of when the USB failure occurs. Please discuss USB / hardware-connect issues on #36

closed 2022-09-16 by timg236

yuyuoo, 2023-04-12:
I am facing the same problem.How to fix it?

trimixtimo, 2023-04-12:
We just use a externally powered USB-switch between PC and CM4 and then it works fine every time

timg236, 2023-04-12:
Locked as duplicate of #36