raspberrypi/usbboot #160: When using the latest usbboot, it fails writing an EEPROM

capiman, 2022-09-21:
C:\Martin\usbboot-latest-uart-enabled\usbboot>"C:\Program Files (x86)\Raspberry Pi\rpiboot.exe" -d recovery
Loading: recovery/bootcode4.bin
Waiting for BCM2835/6/7/2711...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...
Loading: recovery/bootcode4.bin
Sending bootcode.bin
Failed to write correct length, returned 0
Waiting for BCM2835/6/7/2711...

C:\Martin\usbboot-latest-uart-enabled\usbboot>

Notice: I just found out, that bootcode4.bin seems to be empty in my code cloned from github.
The version from 15.07.2022 had a file with a size around 86 KBytes.

Perhaps usage problem: Do I need any make or compile step, before I can call update-pieeprom.sh in recovery?
I have not done anything, beside going to recovery, updating boot.conf and calling update-pieeprom.sh on a system with a CM4.

pelwell, 2022-09-21:
The README explains: https://github.com/raspberrypi/usbboot/blob/master/recovery/README.md

capiman, 2022-09-21:
[image]

pelwell, 2022-09-21:
Do symlinks work on a modern Windows? Try deleting it and replacing it with a copy of the `recovery.bin` from the parent directory.

capiman, 2022-09-21:
First I was using git clone on a linux machine (on a CM4) and tar the complete directory
and transferred it to Windows. I am not sure (or can't remember), if update-pieeprom.sh is also able to run on Windows.
During archiving and extracting the links get lost. Ok, too special, I am doing problematic things, don't matter.

But now I tried to git clone directly under Windows:

C:\Martin>git clone -b master https://github.com/raspberrypi/usbboot.git
Cloning into 'usbboot'...
remote: Enumerating objects: 903, done.
remote: Counting objects: 100% (113/113), done.
remote: Compressing objects: 100% (63/63), done.
remote: Total 903 (delta 55), reused 101 (delta 48), pack-reused 790
Receiving objects: 100% (903/903), 242.58 MiB | 3.72 MiB/s, done.
Resolving deltas: 100% (482/482), done.

C:\Martin>

But then

C:\Martin>dir usbboot\recovery\bootcode4.bin
 Volume in Laufwerk C: hat keine Bezeichnung.
 Volumeseriennummer: 5C63-88DE

 Verzeichnis von C:\Martin\usbboot\recovery

21.09.2022  21:02                15 bootcode4.bin
               1 Datei(en),             15 Bytes
               0 Verzeichnis(se), 39.052.496.896 Bytes frei

C:\Martin>type usbboot\recovery\bootcode4.bin
../recovery.bin
C:\Martin>

It is not more a link, but it is a file containing the destination in ASCII.
The link is broken.
So since introducing the links by @timg236, I assume this

https://github.com/raspberrypi/usbboot/commit/a490517d0b1fca5b369080c3d9ca43d3f59786c9

the way to work with on Windows seems to have changed or is simply not more working anymore...?

capiman, 2022-09-21:
To answer your question: I was able to copy the file of the link from upper directory (like the symlink is doing),
and then flashing the CM4 EEPROM worked again.

capiman, 2022-09-21:
https://superuser.com/questions/1465200/symbolic-link-issue-extracting-tarball-on-windows

timg236, 2022-09-22:
If you want to build this for windows then you'll need Cygwin to build this

timg236, 2022-10-26:
The Readme now indicates that this should be cloned using Cygwin

closed 2022-10-26 by timg236