raspberrypi/usbboot #164: Secure booting cm4 using latest Raspberry PI OS Lite bricks the cm4

ajaykarthikr, 2022-11-02:
Hi,
I was trying to secure boot my cm4 and the documentation. Secure boot till OTP was working and booting up. By after setting `program_pubkey=1` and `revoke_devkey=1` and flashing it. It never booted again, the ACT led did not even blink. There's no HDMI output after this.

I am detailing the steps I followed exactly to provide clarity.

1. Cloned the Raspberry Pi OS boot files
    ```bash
    mkdir secure-boot-files

    cp -R /boot/* secure-boot-files 
    ```
    contents of `cmdline.txt`
    ```txt
    console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 fsck.repair=yes rootwait
    ```

    relevant contents of `config.txt`
    ```txt
    dtoverlay=vc4-fkms-v3d
    dtdebug=1
    ```

2. Then generated boot.img and boot.sig using the following cmds
    ```bash
     sudo tools/make-boot-image -d secure-boot-files -o boot.img -b cm4 -a 64
     
     tools/rpi-eeprom-digest -i boot.img -o boot.sig -k $KEY_FILE
    ```

3. Then copied `boot.img` & `boot.sig` into /boot of CM4

4. Updated the EEPROM using via usbboot.
    ```bash
    cd secure-boot-recovery
    ../tools/update-pieeprom.sh -k $KEY_FILE
    ```
    
    Then set nRPIBOOT to 1 and EEPROM WP protection to zero, and ran the following command.
    ```bash
    sudo ../rpiboot -d .
    ```
    Then, I turned on cm4, it booted up correctly after verifying the boot.sig. Things were going great till this step.

5. To enable OTP, edited `secure-boot-recovery/config.txt` and added
    ```bash
    program_pubkey=1
    revoke_devkey=1
    ```

    After this powering on cm4 doesn't work, there's no led blinking. The ACT light turns on never but never blinks.

Then I tried the recovery methods, ran this from secure-boot-recovery directory itself. 
```bash
 cd secure-boot-recovery
 ../tools/update-pieeprom.sh -k "${KEY_FILE}"
sudo ../rpiboot -d .
```
It still it did not boot up. 

But I was able to get `secure-boot-msd` working but the cm4 never booted when RPIBOOT was set to 0. Is this board bricked?

Is there any mistake in the steps I missed or followed. Some good documentation will help.

Note: I used same private key everywhere which I generated during start.

timg236, 2022-11-02:
This is a duplicate of a forum post that I’ve already answered. It does not brick
your pi and has no way of changing the power led. Closing

timg236, 2022-11-04:
If secure-boot-msd is running it's not bricked since it's able to load and run code. Can you try the previous version of pieeprom.bin works better when booting from MMC?
https://github.com/raspberrypi/usbboot/commit/5b9d1057bbcf9c4fbcb92b031ea7b383fb491f5b

ajaykarthikr, 2022-11-07:
Hi @timg236 I will tried the latest changes, it worked well. I will try to repeat the process with a different PI. I will let you know In case there are any issues.

timg236, 2022-11-09:
Closing because the MMC secure-boot fix was confirmed here
https://forums.raspberrypi.com/viewtopic.php?p=2051770#p2051917

closed 2022-11-09 by timg236