raspberrypi/usbboot #173: RPI boot not working on windows and Mac for CM4 rev5

mezarkabul, 2022-12-09:
### Describe the bug

I have 3 version of cm4.
First one, 
Marking 2921 5.4 Raspberry Pi 2020(before rev5)
It is 8GB ROM 1GB RAM Wifi version.
I can upload image without any problem on any platform, rpi boot working on any platform.

Second one,
Marking 1522 3.1 Raspberry Pi 2021(rev5)
It is 32GB ROM 4GB RAM Wifi version.
I can upload image on linux with rpi boot and imager. 
But when i try on windows rpi boot not work, freezing at "File read: start4.elf" step if cm4 is empty.
If cm4 has a image on eMMc(uploaded on linux), rpi boot waiting about 10 second at "File read: start4.elf" step then finishes and shut down. After 30 seconds computers usb ports stoped work because cm4, cm4 killing all port. When i eject usb cable ports work correctly.
When i try with mac m1 it freezing "Loading embedded: bootcode4.bin"

Third one, 
Marking 1222 4.5 Raspberry Pi 2021(rev5)
It is 16GB ROM 2GB RAM Wifi version. 
It has same problem as second. 

[image]
[image]
[image]
[image]

### Steps to reproduce the behaviour

I have a lot of cm4 and diffrerent io board. Rev5 products gave the same result every time

### Device(s)

Raspberry Pi CM4 Revision 5

timg236, 2022-12-10:
I used Windows (10) with R5 this week and it seemed ok. Please could you trying installing the latest release? This should include a new icon for the mass-storage-gadget. That’s also worth a try and should be faster writing EMMC.

The screenshot for Windows looked a bit old but it’s hard to tell for sure eg the version number might have scrolled off screen

mezarkabul, 2022-12-12:
I am using stable version and last version in my try. There is only one release. Also i am using balena etcher, nothing work..

mezarkabul, 2022-12-12:
Hi again, i install 26 oct 2022 version again(win10 and 11). I dont know what is changed but it is work now. Balena still not work.

timg236, 2022-12-12:
The rpiboot firmware is embedded in the start*.elf and bootcode*bin files. Everything else is the open source c code. Since the new hardware revision has a new PMIC if old firmware is used then it will fail.

In the original Windows screenshot I didn't see a version string. I would expect that it now displays

```
RPIBOOT: build-date Oct 26 2022 version 20220815~145439 9e57f462
Waiting for BCM2835/6/7/2711...
```

I think Etcher either embeds an old version of the software or does some incorrect matching of USB descriptors. That's out of our control but the source code for rpiboot + rpi imager is freely available under BSD license and can be embedded in commercial products if necessary.

timg236, 2022-12-15:
Closing this as resolved since this is working on Windows and Linux.
N.B. We don't maintain the Mac version or Etcher.

closed 2022-12-15 by timg236